cncjs/cncjs #538: Machine Coordinates Not Updating in DRO

swpaxton66, 2019-12-10:
#### Description

Starting with your latest release destop app 64bit for Windows and working my way backwards in versions I cannot get any of them to work properly.

I am running tinyGV8 controller.

The machine coordiantes in the DRO do not update.  They are stuck at zero for all Axis

#### Versions
- CNCjs: 1.9.20, 1.9.19, 1.9.18, etc...
#### How Do You Install CNCjs?
- [x] Download the CNCjs Desktop Application
#### CNC Controller
TinyG V*
#### Hardware
- [x] Desktop or Laptop
#### Operating System
- [x] Windows

MitchBradley, 2019-12-10:
I don't have a TinyG controller to test, but I do have a g2core one.  They are similar but g2core has diverged a little from TinyG so my results might not be entirely comparable.  That said, here is something to try that might give some clues.

Starting with the connection to TinyG closed, use the Connection widget to open a connection to the serial port in TinyG mode.

When the connection is established, the Console widget should expand to reveal a white-on-black area that shows the data that is exchanged with the TinyG controller.  Some of the lines will be too long to fit in the window, but you can scroll horizontally to see them or use the Enter Full Screen button at the top of that widget to make the widget larger.  You can also select and copy all of the text using the mouse, then paste it into a reply on this thread.

Here is what I get:

```
CNCjs 1.9.20 [TinyG]
Connected to /dev/ttyACM0 with a baud rate of 115200
{"r":{"fv":0.99,"fb":101.03,"fbs":"101.03-17-g4cd0e-dirty","fbc":"settings_gShield_Taig_hybrid.h","hp":"ArduinoDue","hv":"na","id":"0084-d639-0084-9ce","msg":"SYSTEM READY"},"f":[1,0,1]}
feeder> {ej:1}
feeder> {jv:4}
feeder> {qv:1}
feeder> {sv:1}
feeder> {si:100}
{"r":{"ej":1},"f":[1,0,8]}
{"r":{"jv":4},"f":[1,0,8]}
{"r":{"qv":1},"f":[1,0,8]}
{"r":{"sv":1},"f":[1,0,8]}
{"r":{"si":100},"f":[1,0,10]}
feeder> {spe:n}
{"r":{"spe":null},"f":[1,100,9]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
feeder> {spd:n}
{"r":{"spd":null},"f":[1,100,9]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
{"sr":{"line":0,"posx":0,"posy":0,"posz":0,"posa":0,"feed":0,"vel":0,"unit":1,"coor":1,"dist":0,"admo":1,"frmo":1,"momo":4,"stat":1}}
feeder> {spc:n}
{"r":{"spc":0},"f":[1,0,9]}
feeder> {sps:n}
{"r":{"sps":0},"f":[1,0,9]}
feeder> {com:n}
{"r":{"com":0},"f":[1,0,9]}
feeder> {cof:n}
{"r":{"cof":0},"f":[1,0,9]}
feeder> {sr:{stat:t,line:t,vel:t,feed:t,unit:t,coor:t,momo:t,plan:t,path:t,dist:t,admo:t,frmo:t,tool:t,posx:t,posy:t,posz:t,posa:t,posb:t,posc:t,mpox:t,mpoy:t,mpoz:t,mpoa:t,mpob:t,mpoc:t,spc:t,sps:t,com:t,cof:t}}
feeder> {sys:n}
feeder> {mt:n}
feeder> {pwr:n}
feeder> {qr:n}
feeder> {sr:n}
{"r":{"sr":{"stat":1,"line":0,"vel":0,"feed":0,"unit":1,"coor":1,"momo":4,"plan":0,"path":2,"dist":0,"admo":1,"frmo":1,"tool":0,"posx":0,"posy":0,"posz":0,"posa":0,"posb":0,"posc":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0,"mpob":0,"mpoc":0,"spc":0,"sps":0,"com":0,"cof":0}},"f":[1,0,206]}
{"r":{"sys":{"fb":101.03,"fv":0.99,"fbs":"101.03-17-g4cd0e-dirty","fbc":"settings_gShield_Taig_hybrid.h","hp":"ArduinoDue","hv":"na","id":"0084-d639-0084-9ce","jt":0.75,"ct":0.01,"zl":0,"sl":1,"lim":0,"saf":1,"m48":1,"froe":0,"fro":1,"troe":0,"tro":1,"mt":2,"tv":1,"ej":1,"jv":4,"qv":1,"sv":1,"si":100,"gpl":0,"gun":1,"gco":1,"gpa":2,"gdi":0}},"f":[1,0,9]}
{"r":{"mt":2},"f":[1,0,8]}
{"r":{"pwr":{"1":1,"2":1,"3":1,"4":1,"5":1,"6":0}},"f":[1,0,9]}
{"r":{"qr":48},"f":[1,0,8]}
{"r":{"sr":{"stat":1,"line":0,"vel":0,"feed":0,"unit":1,"coor":1,"momo":4,"plan":0,"path":2,"dist":0,"admo":1,"frmo":1,"tool":0,"posx":0,"posy":0,"posz":0,"posa":0,"posb":0,"posc":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0,"mpob":0,"mpoc":0,"spc":0,"sps":0,"com":0,"cof":0}},"f":[1,0,8]}
{"sr":{"stat":1,"unit":1,"coor":1,"momo":4,"plan":0,"path":2,"tool":0,"posx":0}}
>
```

The next step is to type the character '?' followed by Enter at the > prompt and cut/paste that result.  Here's mine:

```
> ?
{"r":{"sr":{"stat":1,"line":0,"vel":0,"feed":0,"unit":1,"coor":1,"momo":4,"plan":0,"path":2,"dist":0,"admo":1,"frmo":1,"tool":0,"posx":0,"posy":0,"posz":0,"posa":0,"posb":0,"posc":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0,"mpob":0,"mpoc":0,"spc":0,"sps":0,"com":0,"cof":0}},"f":[1,0,2]}
>
```

As you can see, machine coordinates are present in the "sr" status report, indicated by the "mpox", "mpoy", and "mpoz" fields.

Then you can jog to a different position and watch the status reports show the position as it changes.

```
feeder> G91
{"r":{},"f":[1,0,4]}
feeder> G0 X1
{"r":{},"f":[1,0,6]}
{"qr":47}
feeder> G90
{"r":{},"f":[1,0,4]}
{"sr":{"stat":5,"vel":105.52,"momo":0,"dist":1,"posx":0.02332,"mpox":0.02614}}
{"sr":{"vel":463.08,"posx":0.63874,"mpox":0.63874}}
{"sr":{"vel":2.73,"posx":0.9999,"mpox":0.9999}}
{"qr":48}
{"sr":{"stat":3,"vel":0,"dist":0,"posx":1,"mpox":1}}
```

As you can see, in my system the status reports show dynamically updated X positions in both work coordinates and machine coordinates.  Only values that have changed since the last report are included.

Finally, you can get a full report by typing '?' as before.

```
> ?
{"r":{"sr":{"stat":3,"line":0,"vel":0,"feed":0,"unit":1,"coor":1,"momo":0,"plan":0,"path":2,"dist":0,"admo":1,"frmo":1,"tool":0,"posx":1,"posy":0,"posz":0,"posa":0,"posb":0,"posc":0,"mpox":1,"mpoy":0,"mpoz":0,"mpoa":0,"mpob":0,"mpoc":0,"spc":0,"sps":0,"com":0,"cof":0}},"f":[1,0,2]}
>
```

If you will follow this procedure and report the results, I may be able to deduce what is going wrong.

swpaxton66, 2019-12-11:
Hi Mitch,

Thanks for your help.

Something seems off

Here is the connection info:

NCjs 1.9.20 [TinyG]
Connected to COM3 with a baud rate of 115200
feeder> {ej:1}
feeder> {jv:4}
feeder> {qv:1}
feeder> {sv:1}
feeder> {si:100}
{"r":{"ej":1},"f":[1,0,7,2003]}
{"r":{"jv":4},"f":[1,0,7,7333]}
{"r":{"qv":1},"f":[1,0,7,6003]}
{"r":{"sv":1},"f":[1,0,7,9250]}
{"r":{"si":100},"f":[1,0,9,2707]}
feeder> {spe:n}
{"r":{"spe":null},"f":[1,100,8,8836]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
feeder> {spd:n}
{"r":{"spd":null},"f":[1,100,8,6786]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
feeder> {spc:n}
{"r":{"spc":null},"f":[1,100,8,4736]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
feeder> {sps:n}
{"r":{"sps":null},"f":[1,100,8,704]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
feeder> {com:n}
{"r":{"com":null},"f":[1,100,8,2342]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
feeder> {cof:n}
{"r":{"cof":null},"f":[1,100,8,4826]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
feeder> {sr:{stat:t,line:t,vel:t,feed:t,unit:t,coor:t,momo:t,plan:t,path:t,dist:t,admo:t,frmo:t,tool:t,posx:t,posy:t,posz:t,posa:t,posb:t,posc:t,mpox:t,mpoy:t,mpoz:t,mpoa:t,mpob:t,mpoc:t}}
feeder> {sys:n}
feeder> {mt:n}
feeder> {pwr:n}
feeder> {qr:n}
feeder> {sr:n}
{"r":{"sr":{"stat":true,"line":true,"vel":true,"feed":true,"unit":true,"coor":true,"momo":true,"plan":true,"path":true,"dist":true,"admo":true}},"f":[1,100,181,722]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
{"r":{"sys":{"fb":440.20,"fv":0.970,"hp":1,"hv":8,"id":"3X3566-YQX","ja":100000,"ct":0.0100,"sl":0,"st":0,"mt":10.00,"ej":1,"jv":4,"js":1,"tv":1,"qv":1,"sv":1,"si":100,"ec":0,"ee":0,"ex":1,"baud":5,"net":0,"gpl":0,"gun":1,"gco":1,"gpa":2,"gdi":0}},"f":[1,0,8,5809]}
{"r":{"mt":10.00},"f":[1,0,7,4004]}
{"r":{"pwr":{"1":0,"2":0,"3":0,"4":0}},"f":[1,0,8,9987]}
{"r":{"qr":32},"f":[1,0,7,7766]}
{"r":{"sr":{"posx":0.000,"posy":0.000,"posz":3.000,"posa":0.000,"feed":0.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":3}},"f":[1,0,7,1336]}
>

swpaxton66, 2019-12-11:
and there is the report (?):

X position:          0.000 mm
Y position:          0.000 mm
Z position:          3.000 mm
A position:          0.000 deg
Feed rate:           0.000 mm/min
Velocity:            0.000 mm/min
Units:               G21 - millimeter mode
Coordinate system:   G54 - coordinate system 1
Distance mode:       G90 - absolute distance mode
Feed rate mode:      G94 - units-per-minute mode (i.e. feedrate mode)
Machine state:       Stop
tinyg [mm] ok>

MitchBradley, 2019-12-11:
I think the problem may be related to line length limitations in the TinyG input buffer.  The command that sets the status report format {sr:{stat:t,line:t,vel:t,feed:t,unit:t,coor:t,momo:t,plan:t,path:t,dist:t,admo:t,frmo:t,tool:t,posx:t,posy:t,posz:t,posa:t,posb:t,posc:t,mpox:t,mpoy:t,mpoz:t,mpoa:t,mpob:t,mpoc:t}} is pretty long and it is probably overflowing the buffer.

I am working with the TinyG developers on a private chat group to see what we can do about it.

MitchBradley, 2019-12-11:
I have something for you to try.

Go to the folder  /Users/YOURNAME/AppData/Local/Programs/cncjs-app/resources/app/app/ and use a text editor to edit the file index.js .  Search for the string  `admo:`  You should see a line like:

`        admo: true, // arc distance mode`

Replace "true" with "false" on that line.  Save the file and run cncjs.  Tell me if it works.

swpaxton66, 2019-12-12:
Thanks Mitch.  That did the trick.  It is working now.  Did I have to rob Peter to pay Paul?  (Did I lose any functionality?)

As a side note, how did you get your hands on a g2core?  I see they aren't selling them anymore.

It would be nice if I could talk Cheton into implementing a "programmatic" feed rate override and inject a feed multiplier / divider into the buffer stream :-)

MitchBradley, 2019-12-13:
You lost nothing. TinyGV8 does not support arc distance mode so turning off the report is fine.

G2core is firmware, not hardware that can be purchased. I run it on an Arduino Due connected to a gShield board.  Both can be readily purchased.

cheton, 2019-12-13:
We can do feature detection (e.g., just like `spe`, `spd`) to determine whether `admo` is supported before issuing a status query command. I’m not sure if `tool` (active tool report) is also available in TinyG firmware, we might need to detect the availability if it is not supported in earlier releases.

[image]

MitchBradley, 2019-12-13:
"tool" is available in the TinyG firmware released on TinyGV8, but it is unclear if it is there on TinyGV7.

Instead of hardcoding the list of probed features, I was thinking of initializing the ones to probe with false, then doing something like this:
```
for (var key in sr) {
  if (sr.hasOwnProperty(key) && sr.[key] === false) {
       sr[key] = true;
       await delay(100);
       send(`{${sr[key]}:n}`);
  }
}
```
We would need some way to make the code like this generic:
```
            if (r && r.spe === null) {
                this.sr.spe = false; // No spindle enable
            }
```
Perhaps the for loop could set a variable to indicate which feature is being probed.

Inserer-Pseudo, 2020-10-29:
Hello,
I've got almost the same problem. Only the work position have a feedback.
[image]

I tried to find "index.js" with no success and no trace of "admo=true", see picture
[image]

I found this "index.js" and tried to set "admo" to false but nothing changed, it was under ressources/app/server
[image]
I'm running the latest version of cncjs with a tinygV8

Thanks for your time