cncjs/cncjs #539: Request - query pin states in macros

MitchBradley, 2019-12-12:
This feature request comes from a user in the Facebook CNCjs User's Group.  The request was written in Turkish and machine translated, so I might not understand it correctly.  I think that the request is to be able to check pin state in a macro and then conditionally execute code depending on that state.

The problems of synchronizing with the machine might make this feature difficult to implement reliably.

hsyngndz85, 2019-12-12:
Takim degistirmek icin gercek prosedur şu şekilde olmali. M6 komutu alindiginda freze durmali ve frezeden frekansin 0 hz oldugu bilgisi alinmali. Frekansin 0 hz oldugu bilgisi alindiginda takim tutucudaki yerine birakilmali kilidin acildigi bilgisi kontrol edilmeli. Sonra yeni takim tutulmali kilidin kapandigi gorulmeli sonra isleme devam edilmeli aksi halde cok tehlikeli kazalar olusur. Bu is ic8n grblde 2 pine ihtiyac var ve cncjs de de bu 2 pinin kontrolu gerekli

MitchBradley, 2019-12-12:
Google translation of the above:

The actual procedure for changing the tool must be as follows. When M6 command is received, the milling must stop and the frequency should be informed that the frequency is 0 speed. When the frequency is 0 speed, it should be checked that the lock is unlocked. Then the new tool must be kept locked, the process must be continued, otherwise dangerous accidents will occur. This job needs 2 pins gr8de and cncjs also need control of these 2 pins

MitchBradley, 2019-12-12:
The edge version of g2core has automatic spindle control that will stop the spindle when GCode execution is paused.  I'm not sure how well GRBL handles the situation.