cncjs/cncjs #540: Easy installation ideas

MitchBradley, 2019-12-13:
Many new users are experiencing installation problems on systems where they must perform an "npm install".  Users who can install the electron app have less difficulty since there is an all-inclusive package, but the ones who must install via npm have a lot of trouble.  There are many steps that generate lots of output that is hard for non-programmers to understand.

The serialport module is the most common source of trouble.  Sometimes it must be compiled, which requires a huge compiler toolchain that often fails to install correctly on Windows.  In other cases, a precompiled binary is present, but it is the wrong version due to the problem of having to downgrade the version for the electron app. Even when it does compile correctly, there are a lot of warnings that confuse people.

Even when everything works, the installation can take a long time due to all the stuff that npm must download.

I think we could make a much smaller image that works everywhere.  The idea is to select only those files that are needed at runtime - omitting the many modules that are only used during compilation and testing - and add precompiled serialport binaries for the most important systems. I have tried the "selection" part of this and it works very well.  The resulting tarball is small - 10 MB expanding to 63 MB, compared to the 223 MB uncompressed size of an npm installation.

I think something like this would make it much easier for new users to adopt cncjs, and easier to upgrade to new releases.

tawera, 2019-12-13:
I remember installing cncjs on an RPI about a year ago and it was a very confusing endeavour. Not because it was actually difficult, but because the installation instructions contained a lot of unnecessary stuff and things copy/pasted from related projects, and the stuff that was there contained several alternatives including code that was commented out in confusing ways. 

If I remember correctly, the installation was actually a one-liner:

`$ npm install -g cncjs`

I.e. the one you would expect for any js-application. Whereas if you followed the instructions you would get into all sorts of problems and had to learn about a lot of unnecessary stuff. I think this is a big hurdle for new potential users. Much of this may have been fixed since then but my guess is that much would be gained by keeping things simple instead of trying to provide alternatives for example.

I'm not sure I'm contributing anything useful here, sorry for that. My guess though is that the space and time aspect is not such a big issue for users, and the pure javascript way of doing things has the big advantage of being platform agnostic and low maintenance. A thing that many software projects do is provide a shell script for the installation, giving users a one-liner for installing their software, as for example nvm do:

`curl -o- https://raw.githubusercontent.com/nvm-sh/nvm/v0.35.1/install.sh | bash`

This could be a solution for hiding all the confusing output of npm for example, and doing other stuff, such as setting up autostart maybe.

saxa, 2019-12-13:
For sure the install script or a kind of installer is the way to go.

MitchBradley, 2019-12-13:
@tawera Read the current version of https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Install-Node.js-via-Package-Manager-*(Recommended)* .  It dispenses with the alternatives.

You cannot simply type "npm install -g cncjs" because nodejs is not guaranteed to be installed already, and if it is installed the version might be wrong, and if it needs to recompile the serialport module, a toolchain must be present.

There are some things that can go wrong even if you follow the simplified instructions. For example, the full npm install process can overwhelm some platforms that could otherwise run CNCjs in some fashion - such as the Pi Zero W that runs my production mill. A script reduces the effort of typing the individual commands.  That is a good thing, but at the expense of hiding the steps so if something goes wrong, it is harder to untangle the effort.

My suggestion is based on my recent experience supporting new users.  I am currently the only one who appears to be doing so at the level of actively engaging with them and working through problems. There are several people who, in response to questions on the Facebook group, give pointers to YouTube videos that often do not answer the question at all, but few who are supporting at the level of "tell me exactly what you did, step by step, and we will work through it until you succeed".

That support is another factor that makes me long for a faster procedure.  When you are working with somebody one-on-one, any step that takes a long time increases the probability that one of us will have to disengage due to some other appointment. That often means having to re-engage the next day.  A support process that could otherwise be completed in one short session can take days, which is very frustrating for everyone concerned.  Especially me.

tawera, 2019-12-14:
So long as the installation procedure is reliable I don't think the time and space issue is critical. But I agree, because it isn't, it's a pain point. And the npm/building from scratch-approach is never going to be the most reliable.

I'm not a javascript developer so I don't know what I'm talking about, but maybe Docker would actually be the best solution for cncjs. 

The downside is that docker is one more component that needs to be added to the mix, but it provides the separation of concerns needed. The problem is making cncjs run in all imaginable environmnents, add in compiling non javascript-modules and this makes it incredibly comlex. Using docker reduces the installation to two problems. One, creating a docker image, I believe there already is one. And two, installing docker on your system, this requires no effort on cncjs part. 

I'm sure you have looked into docker already, but just maybe it would be the best way to go forward, once you have docker on your system, cncjs should be pretty much bombproof and you can run the same image anywhere. There still is the matter of mapping the http port, serial device and config file to the host. And I'm not entirely sure about the story with windows and docker.

I believe docker would take care of autostart, exposing port 80, and updating cncjs nicely as well and it would work the same on all platforms, and it should require the least amount of effort on cncjs' part. Of course there's never a silver bullet but this might be something to consider.

MitchBradley, 2019-12-14:
@tawera   Docker is not nearly as trouble-free as one would hope.  For example, if you want cncjs to run on low-end Pi's, you need a different OS image as the base for the docker image, because the ARM architecture differs between Pi versions.  So instead of just having the one native code file (serialport) to worry about, you now need to deal with entire architecture-dependent OS images. And the user has to learn the vagaries of Docker administration, which is not straightforward - the differences between images and containers and what happens to lingering stopped containers and how to persist changes to the image and how to access host resources like files and blah blah blah.

I use Docker on Windows because it is the only route to running some 32-bit compilers that I need, and it is a pain.

I tried to build a Docker image that would run cncjs on a low-end Pi.  I spent several days on the project and never found a satisfactory solution.

I would like to reduce complexity and image size, not multiply it.

The one thing that Docker buys you is the ability to bundle cncjs with a compatible version of nodejs. That advantage comes at the price of numerous disadvantages.

Nodejs itself is a "virtualization layer", in the sense that, with the sole exception of the serialport binary, the entirety of the cncjs build image is machine-independent.  If you do the install process once, you end up with a collection of text files that can run anywhere.  3/4 of the install footprint is no longer needed, as it was used to transform the CNCjs source code into a compact form and do various preparation and testing steps.  The remaining 1/4 can be copied from system to system quite simply.

saxa, 2019-12-14:
> A script reduces the effort of typing the individual commands. That is a good thing, but at the expense of hiding the steps so if something goes wrong, it is harder to untangle the effort.

This is easily solved if you put the correct steps and the correct checking, especially tests for succeeded step.

tawera, 2019-12-15:
Well you need one compiled version of serialport for each supported architecture, and in docker world you need one image per supported architecture. The fact that the whole system comes along for the ride in the docker image, I don't see that as adding complexity. And the fact that you need to separate the mutable and the immutable part of your system is a nice feature, not an unfortunate consequence.

I use docker daily at work, both for development purposes and production systems and for me it's the bees knees because it eliminates this whole class of problems to do with system dependencies and I can run any versions of any dockerized software I want without them interfering with each other. You define the image once and it works the same no matter in which docker host it executes, with performance indistinguishable from running it natively, and it will continue to work when you upgrade the host and no matter what other software you run on the host.

But I agree that it's in no way certain that it would be the best approach for this problem and you have many valid points and have obviously already explored this path. If you can build cncjs and create a tarball or similar that will execute on any machine with a supported node version, that sounds like a great plan to me. I got involved in this discussion because I felt it was a shame that the installation exeperience was a bit of a hassle for this great piece of software, and it would be awesome to see it improve.

Great work on the new setup guide BTW! When I set up my RPI3 I think the first step of the instructions was to install nodejs 7, which failed for me because raspbian already came with nodejs preinstalled, so there was a conflict. I thought I better follow the instructions and use nodejs 7, but I could not uninstall the existing nodejs version because other system packages were dependent on them, so this was the first hurdle. This was all unnecessary in the end of course but I didn't know at the time, because the nodejs version preinstalled was newer and also supported by cncjs. I notice the new guide instructs to install nodejs 8, are you sure this is necessary? And it doesn't state what is the starting point, ie what distribution should you install.

MitchBradley, 2019-12-15:
I don't know what the minimum version of nodejs is.  It's rather time-consuming to uninstall everything and then reinstall everything to see what works. I'm getting pretty tired of the whole process.  If you would be willing to help by creating the map of what works and what doesn't, that would be great.  I *think* that v8 and v10 work, but v12 doesn't.

The number of scenarios is pretty daunting, considering that you can have any preexisting version of nodejs - or none - and any preexisting version of cncjs.  Yes, I realize that is an argument for Docker, but substitute Docker for nodejs above and you have essentially the same problem.

Docker images are portable to machines with the same architecture, or machines with enough grunt to virtualize that architecture, but good luck running a Docker image built on an AMD64 base OS on a low-end Pi, for example.

tawera, 2019-12-16:
One way around these issues would be to provide proper packages for cncjs on raspbian, that should be the most user friendly and most stable solution. But that would also only be a solution for raspbian, and it would require someone creating and maintaining these packages.

I would imagine the README is correct in that versions 6, 8 and 10 will work.

There are currently three different installation images of raspbian

- Raspbian Buster Lite
- Raspbian Buster with desktop
- Raspbian Buster with desktop and recommended software

My guess is that Lite does not come with nodejs while the others do. I don't have a pi available at the moment to test. Either way, if there are nodejs packages in the official repo, they should be available with one apt install which should bring the default version, which hopefully is one that works with cncjs. That should be the most robust way to install node, and if you do apt install and already have node, then it just does nothing.

MitchBradley, 2020-01-01:
Here is an example of the problems that can be encountered when installing cncjs via npm on a Raspberry Pi.  Yes I know this is not the suggested sequence - my aim was to try to do things in the most straightforward way possible, to see what would happen.

We start with the Raspbian Buster Lite image on a Pi Zero W, then

sudo apt install nodejs

That cranks away for awhile, then fails to get libssl.  Oops, forgot to do ``apt update``.  So

sudo apt update
sudo apt install nodejs

Worked this time.  So far so good.  But installing nodejs does not install npm, so we have to install it too:

sudo apt install npm

Then we test npm 

npm -v
5.8.0

Seems a little old.  Let's try it:

$ npm
npm WARN npm npm does not support Node.js v10.15.2
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/

Apparently the packaged versions of nodejs and npm in Debian Buster are not at the same level.  So

sudo npm install -g npm@latest

That claims to have installed npm@6.13.4, but when you run npm, it still runs 5.8.0.  WTF?  Perhaps the packaged npm binary goes in a different directory that is earlier in the search path?

Yep.  Okay, this is super weird:

pi@pi0:~ $ npm
npm WARN npm npm does not support Node.js v10.15.2
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/

Usage: npm <command>

where <command> is one of:
    access, adduser, bin, bugs, c, cache, ci, completion,
    config, ddp, dedupe, deprecate, dist-tag, docs, doctor,
    edit, explore, get, help, help-search, i, init, install,
    install-test, it, link, list, ln, login, logout, ls,
    outdated, owner, pack, ping, prefix, profile, prune,
    publish, rb, rebuild, repo, restart, root, run, run-script,
    s, se, search, set, shrinkwrap, star, stars, start, stop, t,
    team, test, token, tst, un, uninstall, unpublish, unstar,
    up, update, v, version, view, whoami

npm <command> -h     quick help on <command>
npm -l           display full usage info
npm help <term>  search for help on <term>
npm help npm     involved overview

Specify configs in the ini-formatted file:
    /home/pi/.npmrc
or on the command line via: npm <command> --key value
Config info can be viewed via: npm help config

npm@5.8.0 /usr/share/npm
pi@pi0:~ $ which npm
/usr/local/bin/npm
pi@pi0:~ $ ls -l `which npm`
lrwxrwxrwx 1 root root 38 Jan  1 17:54 /usr/local/bin/npm -> ../lib/node_modules/npm/bin/npm-cli.js
pi@pi0:~ $ ls /usr/share/npm/bin/npm-cli.js
/usr/share/npm/bin/npm-cli.js
pi@pi0:~ $ ls -l /usr/share/npm/bin/npm-cli.js
-rwxr-xr-x 1 root root 2749 Mar 23  2018 /usr/share/npm/bin/npm-cli.js
pi@pi0:~ $ /usr/share/npm/bin/npm-cli.js -v
5.8.0
pi@pi0:~ $ /usr/local/lib/node_modules/npm/bin/npm-cli.js
^C
pi@pi0:~ $ /usr/local/lib/node_modules/npm/bin/npm-cli.js  -v
6.13.4
pi@pi0:~ $ `which npm` -v
6.13.4
pi@pi0:~ $ npm -v
5.8.0
pi@pi0:~ $ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
pi@pi0:~ $ /usr/local/sbin/npm
-bash: /usr/local/sbin/npm: No such file or directory
pi@pi0:~ $ /usr/local/bin/npm -v
6.13.4
pi@pi0:~ $ /usr/sbin/npm -v
-bash: /usr/sbin/npm: No such file or directory
pi@pi0:~ $ /usr/bin/npm -v
5.8.0
pi@pi0:~ $ /bin/npm -v
-bash: /bin/npm: No such file or directory
pi@pi0:~ $ /usr/local/games/npm -v
-bash: /usr/local/games/npm: No such file or directory
pi@pi0:~ $ /usr/games/npm -v
-bash: /usr/games/npm: No such file or directory

So how is it that npm is being run from /usr/bin even though the /usr/local/bin version is earlier in the search path?????

Oh, the infamous bash path cache.  I hadn't thought about that in years.  So

pi@pi0:~ $ type npm
npm is hashed (/usr/bin/npm)
pi@pi0:~ $ hash -r
pi@pi0:~ $ npm -v
6.13.4

Imagine a non-programmer being able to solve that problem!

time sudo npm install -g --unsafe-perm cncjs

That is cranking away.  It is taking a really long time on a pi0 and I have to go do something else now, so this will do for now.

tawera, 2020-01-02:
If the distribution-provided node and npm are broken then that is not the way to go, but that is a bit surprising. Maybe the node packages on raspbian are not well maintained, or this is just an anomaly that will be fixed soon?

MitchBradley, 2020-01-02:
It is not at all surprising to me, having used (and worked for companies that distribute) dozens of flavors of Unix and Unix-like systems over 40+ years.  The combinatoric problem of getting every package in a distro to be just right, in the face of version churn of every one of them, with thousands of unaffiliated package authors, is essentially intractable.  Even monolithic companies like Microsoft and Apple, with huge well-funded testing groups and direct control over most of their software, make mistakes.

npm is a hot mess.  "sudo apt install npm" gets 246 sub-packages, including gyp, 5 lib*'s, and 240 fairly small node-*'s. The install process generates about 1000 lines of output, meaningless to a non-programmer. The typical reaction of such a user is "I got lost immediately".  And then when you try to use it, the first message you get is that it is out of date.  So you have to use a different command to update it (sudo npm install -g npm), and that generates an entirely different long set of messages, and installs the new version in a different location that can't be found immediately (because of the bash caching thing). From an app deployment standpoint, this is just a disaster.

Then, when (if ever) you get past the npm installation problem, you use it to install cncjs, only to see an unmanageably large pile of output riddled with scary warnings about stuff that is unsafe and you (meaning the programmer) really should be using a newer version, and appeals for money and jobs, and brightly colored compiler warnings about really obscure C++ syntax issues that didn't even exist until some recent feature was added to C++.  All this is really scary to naive users, and if there is an actual error, chances are that they won't be able to find it among all the other stuff.  Of course a programmer would just grep for errors in the log file, but programmers are not the audience.

I understand the intellectual appeal - and the utility to developers - of the fine-grained modularity represented by the node way of doing things, but it comes with some serious disadvantages when your objective is to deliver an app that "just works" to a "customer".

Let us suppose that this packaging problem will be fixed soon.  Where does that leave me, as someone who wants to help people use cncjs?  Do I tell them they have to wait until it gets fixed (if ever)?  Or do I publish a convoluted recipe that works now, but will probably be wrong later when it does get fixed, and then keep polling the distro to see if it gets fixed?  And then what?  "If you are using version a, then you have to do x, but if the version is b, then you do y, ad infintum".  Users don't like decision trees.

I find it very appealing to be able to say (in the form of the commands to do these steps)

1. install nodejs via the package manager
2. download a smallish tarball and unpack to a specific directory

tawera, 2020-01-02:
Well, that the provided node and npm versions are incompatible I have to imagine would be a critical showstopper issue in any of the mainstream linux distributions but Raspbian may simply not care about node support, I don't know.

I don't think anyone is aguing against your approach. Having node as a single dependency and not having to build everything from scratch would be a big improvement.

saxa, 2020-01-03:
I would say that an installer script is good just because of this kind of things. It can easily make a check for the correct npm installed version and throw an error or even ask to correct it.

MitchBradley, 2020-01-04:
I found a tool [prebuildify ](https://github.com/prebuild/prebuildify) that seems to solve the problem of packaging multiple versions of native modules.  It builds native binaries for different architectures, node ABI versions, and node-vs-electron, putting them in a directory where node-gyp-build can select the correct one for the runtime system.

I also did some installation time studies on a Raspberry Pi Zero W, which is probably the slowest - and lowest cost - system on which CNCjs will run.  The differences are striking:

- apt install nodejs + apt install npm + npm install cncjs + npm rebuild (for serialport) :  about 40 minutes, over 1000 lines of unhelpful output, about 50 warnings, and 3 appeals for donations/jobs
- curl nodesource ... : doesn't work on Pi Zero W because nodesource stopped supporting ArmV6
- wget nodejs.org ... + tar x + npm install cncjs + npm rebuild: about 30 minutes
- wget nodjs.org  ... + tar x + wget cncjs tarball + tar x: about 3 minutes