cncjs/cncjs #541: Request: Macro way to access GrblRunner.parameters

markwal, 2019-12-18:
#### Description

I'd like to be able to reference some of the line parser result parameters from a macro.  In particular, I'd like to do something like:

G90
G38.2 Z-105 F200
%wait%
G43.1 Z[parameters.prb.z - TOOL_SENSOR_HEIGHT]

instead of using mposz or posz since on my Nomad there is a slight difference between the two and the PRB report is the repeatable one.

Perhaps there is already a way to do this, but spelunking around hasn't revealed it to me yet.

#### Versions

- CNCjs: 1.10.0 (built from latest master, commit 4b28f52305, changed a couple of export lines to workaround some i18next messages)
- Node.js: v8.11.2
- NPM: 6.13.4

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

neilferreri, 2019-12-31:
Hey @markwal,
This is interesting. Why are you not using the work position?

markwal, 2019-12-31:
Same problem with machine and work position, the PRB result is more accurate and repeatable.  I'm guessing not by a lot though.