cncjs/cncjs #542: Virtual serial port not recognised

RealTadango, 2019-12-19:
#### Description

CNCjs does not recognise my virtual comport, and has no option to manually specify a com port. I connect to my CNC over Ethernet and use a virtual com port driver for compatibility. It would be nice to have an option for manual selection of com ports. UGS has it and GrblGru also works with this solution. 

#### Versions

Electron app 1.9.20

#### How Do You Install CNCjs?

- [ ] NPM
- [X] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

cheton, 2019-12-19:
So far the configuration is not available on the main UI, you have to edit the `.cncrc` file to manually specify a port.  A configuration example is available at here: https://github.com/cncjs/cncjs/blob/master/examples/.cncrc#L2-L7

RealTadango, 2019-12-19:
@cheton Thanks, i found the .cncrc in my home directory (C:\Users\<UserName>\.cncrc) and added the virtual port and it works :) Thanks!

closed 2019-12-19 by RealTadango