cncjs/cncjs #543: CNCjs not "starting" when connecting to Grbl with only RX / TX lines

RealTadango, 2019-12-19:
#### Description

Normally when CNCjs connect to Grbl over a serial port the port connect resets the controller and Grbl sends an initial response. It looks like CNCjs is not working without the initial response of Grbl. I want to use CNCjs directly over serial without the reset lines connected, so only RX / TX. This is not working since there is no initial response from Grbl when CNC is connecting. How can i get CNCjs to work? Using the terminal i can send all commands normally and get a response.

#### Versions

Electron app 1.9.20

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

RealTadango, 2019-12-19:
Sorry, i found it myself.... Using the reset button triggers it......

closed 2019-12-19 by RealTadango

noname1020, 2022-04-03:
I used to push reset button in order to connect fluidnc with cncjs,
then I figured that I can give a reset signal by enter $Bye (command for reset MCU in fluidnc equivalent to push reset botton) at cncjs console,so make it easier here is how I did it.
Add an a events at setting like this.
Events - Open a serial port (System only)
Trigger - System
Commands - stty -F /dev/ttyUSB0 115200 cs8 -cstopb -parenb -icanon
                      sleep 1
                      echo -e '$Bye' > /dev/ttyUSB0
Save.
Click open botton,no more push physical reset botton.
Hope this helps