cncjs/cncjs #544: Resume after limit switch triggered stop.

pelted, 2019-12-21:
For about the 4th time I’ve bumped one of my limit switches on my LEAD CNC while vacuuming up chips. I can not seem to resume the processes after this and always have to restart the whole cut. Is this a function of Grbl or CNCJS? Is there any way to actually resume after being unlocked when this happens?

imwhocodes, 2019-12-21:
This is a Grbl feature, by hitting the switch you engaged the “hard limit emergency”, you can disable this feature https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#21---hard-limits-boolean but I would advise against that.

neilferreri, 2019-12-21:
It's a grbl thing. There's no way to guarantee position was not lost on a crash into a limit switch.
Your options would be to disable hard limits (good toolpaths don't need them) and just stop hitting the switches while s job is running.  Maybe make/buy a dust shoe.

closed 2022-03-27 by emcniece