cncjs/cncjs #545: SyntaxError: Block-scoped declarations ?

RossosHEX, 2019-12-26:
#### Description

When trying to run cnc from command line I am getting an error.

```
~/project_3D/cnc/cncjs$ cnc -v
Error: [SyntaxError: Block-scoped declarations (let, const, function, class) not yet supported outside strict mode]
~/project_3D/cnc/cncjs$ cnc -h

  Usage: cnc [options]

  Options:

    -V, --version                       output the version number
    -p, --port <port>                   Set listen port (default: 8000) (default: 8000)
    -H, --host <host>                   Set listen address or hostname (default: 0.0.0.0) (default: 0.0.0.0)
    -b, --backlog <backlog>             Set listen backlog (default: 511) (default: 511)
    -c, --config <filename>             Set config file (default: ~/.cncrc)
    -v, --verbose                       Increase the verbosity level (-v, -vv, -vvv)
    -m, --mount <route-path>:<target>   Add a mount point for serving static files (default: )
    -w, --watch-directory <path>        Watch a directory for changes
    --access-token-lifetime <lifetime>  Access token lifetime in seconds or a time span string (default: 30d)
    --allow-remote-access               Allow remote access to the server (default: false)
    --controller <type>                 Specify CNC controller: Grbl|Marlin|Smoothie|TinyG|g2core (default: '') (default: )
    -h, --help                          output usage information

  Examples:

    $ cnc -vv
    $ cnc --mount /pendant:/home/pi/tinyweb
    $ cnc --mount /widget:~+/widget --mount /pendant:~/pendant
    $ cnc --mount /widget:https://cncjs.github.io/cncjs-widget-boilerplate/v1/
    $ cnc --watch-directory /home/pi/watch
    $ cnc --access-token-lifetime 60d  # e.g. 3600, 30m, 12h, 30d
    $ cnc --allow-remote-access
    $ cnc --controller Grbl

~/project_3D/cnc/cncjs$ cnc -V
1.9.20
~/project_3D/cnc/cncjs$ node -v
v4.9.1
~/project_3D/cnc/cncjs$ npm -v
2.15.11
```

The installation was performed following the steps in https://cnc.js.org/docs/installation/

The step ```npm install npm@latest -g``` was gimping the whole installation, omitted in later attempts.
Had to do ```sudo rm -rf ~/.nvm/versions/node/``` several times as well.

#### Versions

- CNCjs: 1.9.20
- Node.js: v4.9.1
- NPM: 1.9.20

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux
Linux 4.15.0-72-generic #81~16.04.1-Ubuntu SMP Tue Nov 26 16:34:21 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

These are the complete set of commands I've been executing:
```#/bin/bash
git clone https://github.com/creationix/nvm.git ~/.nvm
cd ~/.nvm
git checkout `git describe --abbrev=0 --tags`
cd ..
chmod +x ~/.nvm/nvm.sh
. ~/.nvm/nvm.sh

# Add these lines to your ~/.bash_profile, ~/.bashrc, or ~/.profile file to have it automatically sourced upon login:
# export NVM_DIR="$HOME/.nvm"
# [ -s "$NVM_DIR/nvm.sh" ] && . "$NVM_DIR/nvm.sh" # This loads nvm

nvm install node

nvm install 4
nvm use 4

# npm install -g cncjs
npm install -g cncjs@latest --unsafe-perm
```

Any help and/or suggestions are very much appreciated.

MitchBradley, 2019-12-26:
Recent versions of cncjs are know to work on node versions 8 and 10.  It is likely that version 4 is too early.  You could try uninstalling node and installing version 8.

RossosHEX, 2019-12-26:
Thank you, Mitch.

Following your suggestion I changed the ```nvm install 4 && nvm use 4``` to ```nvm install 8 && nvm use 8``` and run the commands again.
After a min it told me something failed. Changed ```npm install -g cncjs@latest --unsafe-perm``` back to ```npm install -g cncjs``` and it compiled with a few warnings:

```
npm WARN deprecated bcrypt-nodejs@0.0.3: bcrypt-nodejs is no longer actively maintained. Please use bcrypt or bcryptjs. See https://github.com/kelektiv/node.bcrypt.js/wiki/bcrypt-vs-brypt.js to learn more about these two options
npm WARN deprecated superagent@3.8.3: Please note that v5.0.1+ of superagent removes User-Agent header by default, therefore you may need to add it yourself (e.g. GitHub blocks requests without a User-Agent header).  This notice will go away with v5.0.2+ once it is released.
npm WARN deprecated core-js@1.2.7: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
npm WARN deprecated core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
npm WARN react-datepicker@1.5.0 requires a peer of react@^16.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN react-datepicker@1.5.0 requires a peer of react-dom@^16.0.0 but none is installed. You must install peer dependencies yourself.
```

It did not want to launch by typing ```cnc``` in the terminal though.
Had to manually dig into ```~/.nvm/versions/node/v8.17.0/bin/cnc``` which happily launched the application. After quitting it and trying again with ```cnc``` it launched fine. Maybe it needed first time run configuration?

Either way, my machine now happily runs cncjs@1.9.20.
Thank you again for a quick response, and Season Greetings!

PS. Someone please touch up on the Installation guide, change 4 to 8.

closed 2019-12-26 by RossosHEX

MitchBradley, 2019-12-26:
I am working on the installation guide.  Maintaining these guides is getting to be a hassle because there are several conflicting versions that are written by different people, and the supported node version is a moving target.

You should start to use the "cncjs" name to start the program.  "cnc" is deprecated and will go away eventually.