cncjs/cncjs #546: Axes widget not showing current position until M114 is manually sent or...

fantasmag0ria, 2019-12-27:
#### Description

...coordinate system is changed (using G54 etc button).

I was always using grbl (where it worked just fine) but recently moved to Marlin 2.0 (bugfix) + SKR 1.3 board because I wanted to give new TMC5160s a try.  

Am I missing some hidden config flag somewhere in Marlin configuration or is it really a bug?

#### Versions

- CNCjs: 1.9.20

#### How Do You Install CNCjs?

- [x] downloaded v1e rPi image
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core
- [x] Marlin

#### Hardware

- [x] Raspberry Pi Zero W
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

MitchBradley, 2019-12-27:
Marlin is problematic because it does not report position automatically.   cncjs can explicitly request position periodically to keep the DROs updated, but that can cause the GCode program to stutter.  So far we have not found a solution that satisfies all use cases. Marlin appears to be optimized for 3D printing use with a directly-attached LCD display instead of being fed via a serial line with a milling style UI.  You can feed it GCode over a serial line, but getting position updates while that is happening is not easy.  Position updates while 3D printing are relatively uninteresting because most moves complete very quickly, unlike milling where long moves at slow feedrates are relatively common.

fantasmag0ria, 2019-12-27:
Thanks, that's kinda what I suspected. 
Is it, maybe, possible to enable the special mode in which cncjs is actively asking for position when no GCODE is run? 
And to not being requested when a job is running? (not to interfere with it)

closed 2020-01-01 by fantasmag0ria