cncjs/cncjs #549: I implemented G10 L20 () in Marlin 2.0...

fantasmag0ria, 2020-01-01:
#### Description

Based on https://github.com/MarlinFirmware/Marlin/issues/14734 
and 
http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G10-L1_

I have basic understanding of GCODE and C++, so if there is anyone willing to take a look and confirm it does what it is supposed to do here's source code.  

most changes at:  ..\Marlin\src\gcode\feature\fwretract\G10_G11.cpp

I also added position reporting on G0.  

tested on SKR 1.3 and seems to be working fine.

[Marlin.zip](https://github.com/cncjs/cncjs/files/4014282/Marlin.zip)
(copy and replace Marlin directory in Marlin 2.0 repository)

#### Versions

- CNCjs: 1.9.20
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Marlin
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi Zero W
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

telleropnul, 2020-01-05:
Very interesting.  Am not in a position to test at the moment but G10 L2(0) support is much appreciated for CNC machine use (LinuxCNC / CNC.js).  Thank you for your contribution!

You mentioned 

> "I also added position reporting on G0."

I am very curious what this involves...this is what the developer of CNC.js had to say about position reports:

Other (non Marlin) controllers have a soft-configurable (does not require recompilation) option to include both machine and work coordinates in their position reports. For example, g2core tags machine coordinates with "mpos:" and work coordinates with "wpos:" in the JSON-formatted reports. The report format has little or nothing to do with the parsing of GCode. In the case of Marlin, there is no easy way to coerce it to automatically issue timely position reports. You have to send a command asking it for a report, and the periodic sending of those commands sometimes causes motion stuttering because the "send a report" command has to be interspersed with the GCode from the file. The lack of automatic position reports can be troublesome for milling, where it is common for a single GCode move to take several seconds - for example a long arc at a low feedrate. With Marlin, the DROs don't update during that move, and might not update until the end of several such moves, depending on how often CNCjs has inserted "send position" commands.

https://github.com/cncjs/cncjs/issues/514

fantasmag0ria, 2020-01-05:
> I am very curious what this involves

Its a different approach. Instead of polling machine position by cncjs I send it from the controller on every G0 (rapid). Its basically one line of code added to Marlin at the end of G0 function. 

I figured you don't usually run jobs with a lot of G0 (while cutting) so It shouldn't influence performance. And knowing machine position while running GCODE is not as useful as while jogging. 

Also when you want to cut using G0 marlin allows to modify feedrate. So you can do it.

telleropnul, 2020-01-08:
Here is what the developer of CNC.js had to say:
https://github.com/cncjs/cncjs/issues/514

V1engineering forum (MPCNC)
https://forum.v1engineering.com/t/marlin-versus-work-coordinates-g10-l20/10959

Any chance you can make sure the positional reports both return machine coordinates using "mpos:" tag and work coordinates using "wpos:" in the JSON-formatted reports similar to g2core ?

fantasmag0ria, 2020-01-08:
>machine coordinates using "mpos:" tag and work coordinates using "wpos:"

do you have some examples or documentation how it should be reported?

fantasmag0ria, 2020-01-08:
Yeah, it wont work. Marlin position parser looks like this:

https://github.com/cncjs/cncjs/blob/master/src/server/controllers/Marlin/MarlinLineParserResultPosition.js

telleropnul, 2020-01-09:
>do you have some examples or documentation how it should be reported?
>Or how it should be reported for CNCjs to understand it correctly

Sorry, I do not.

CNC.js developer [Mitch Bradley](https://github.com/MitchBradley) knows what automated position reports should look like for optimum CNC.js compatibility:
https://github.com/cncjs/cncjs/issues/514

I just don't know how to get his attention nor if he is willing to provide an example...

fantasmag0ria, 2020-01-09:
I guess he will when he will have time. I guess its not his job to answer our questions ;)

MitchBradley, 2020-01-09:
I did not spend much time looking at the changes because they are not presented in the standard form of a pull request (PR) against a specific version.  There is a very good reason for that standard form - it saves time for primary developers, who are often very busy and overwhelmed with requests from many different people.  With a PR, you can easily see exactly what changed, accept/merge it with the push of a button if it is good, comment on specific lines with a single click, etc.

Another meta issue is with the specific "ask" in the original issue posting "can someone take a look at it and confirm that it does what it is supposed to".  Asking CNCjs developers to review Marlin code is out of scope.  As far as I know, there is no overlap between the CNCjs and Marlin development team. There just happens to be some g2core overlap, as I am also a g2core developer, but even so, I would expect requests for g2core code review to go to the g2core site.

But those meta issues aside, it seems to me that the thread has morphed into three separate issues:

- Coordinate system support via G10
- "automatic" position reports (actually a subset of a fully useful solution to that problem)
- work coordinate reporting

(As with the PR meta-issue, when too many issues get conflated, it is hard for people to focus.  When too-busy people are presented with conflated issues, the tendency is to disengage.)

Taking those one at a time:

G10: So what is the CNCjs issue here?  Does CNCjs work with the changed version or not?  If it does, what is the problem?  If not, an issue specifically identifying how to reproduce the problem would be appropriate - but possibly time-consuming for the CNCjs developers to handle because testing it would require us to set up a Marlin compilation environment, compile the patched code, load it onto a board, etc.  That might be worthwhile if this feature becomes standard on Marlin, but until is clear whether that will happen, it's a "one off" situation.

"not really automatic" position reports:  Again, does this work with CNCjs or not?  My guess is that it does work, because CNCjs just deals with position reports whenever they arrive.  There is no notion of "expected" vs "unexpected" reports.  There is some CNCjs code to try and coerce Marlin to send some reports, but on the receiving side, whenever a position report happens to come in, things get updated.  Perhaps the issue should be framed as an announcement?

"work coordinate reporting": This was framed as "could you add mpos: and wpos: tags like g2core" and "what position reports should look like for optimum CNCjs compatibility". As it currently stands, Marlin reports look nothing at all like g2core reports, so trying to shoehorn g2core syntax into Marlin syntax is not workable.  g2core report syntax is highly structured using JSON notation, whereas Marlin is totally ad-hoc.  With a structured report format, you add new features easily because the parser expects everything to be tagged.  Any new tag is simply ignored by older versions of the code, which is possible because the parser can always tell what is a tag and what are the arguments for that tag.  Ad-hoc formats are very tricky to change without risk of breaking old code.  Since the format is unstructured, different UI agents probably use different strategies for recognizing the various different reports and pulling out the different fields.  You must be very careful to design a new report that old parsers won't confuse for a preexisting report - and just adding fields to an old report suffers from the same problem.   To do this safely requires you to consider every different UI agent that purports to support Marlin - not just CNCjs.

And thus perhaps you can see why I am not a fan of Marlin for the use case of external UI control.  Its serial communications framework is poorly designed at a very fundamental level.  For the use case of running a 3D printer via an attached LCD panel, Marlin works brilliantly.  I use it on my 3D printer.  Copy the GCode to a USB stick, bung it in the printer, and let the job run for hours without worrying about whether an attached PC will go to sleep at a bad time.

The answer to "what should they look like for optimum CNCjs compatibility" is this: Compatibility with current releases of CNCjs is hopeless because CNCjs parses Marlin reports according to existing Marlin syntax which does not have work position. For the future, whatever you can convince the Marlin team to do will be okay with CNCjs, under the assumption that the Marlin team will invent something that is unambiguous (otherwise it will cause endless problems for everybody).  The Marlin and g2core report parsers are totally separate so trying to make an amalgam of the two is not helpful. If Marlin were to use g2core style JSON reports, that might be helpful in theory, but the fact that there are so many copies of Marlin out there that use the existing Marlin format makes a new JSON format not helpful in practice.

Sorry if this sounded cranky - but the fact is that there are lots of factors in play and I wanted to lay them out so you understand the depth of the problem.  Maintaining and enhancing complex software is a big problem, and is made much much harder by interactions between multiple separately-developed components.

fantasmag0ria, 2020-01-09:
Thanks for the in-depth reply.

> not presented in the standard form of a pull request

I know, I need to get it done. I come from svn camp and git is weird for me ;)

> specific "ask" in the original issue posting "can someone take a look at it and confirm that it does what it is supposed to"
> as I am also a g2core developer, but even so, I would expect requests for g2core code review to go to the g2core site

Exactly my point, I figured because Marlin is mainly for 3D printing and have no use for G10 I would get better feedback asking this question here. As I mentioned I am a complete noob in this area. 

I only moved to Marlin (from grbl) because I bought some TMC5160s and wanted to check them out.

> Does CNCjs work with the changed version or not?

*I* think it does. 

Regarding automatic position reports: So it looks like there is no simple one way solution. I will probably live with the half-cooked one. Its good enough for me at the moment.

> Sorry if this sounded cranky

You didn't. Thanks!

MitchBradley, 2020-01-10:
git is weird for everybody.

telleropnul, 2020-01-10:
Many thanks (again) Mitch.

"G54"
https://github.com/MarlinFirmware/Marlin/issues/14743
Executing G54 + G92 X0 Y0 until recently would reset G53 machine coordinates origin (**fixed**)

"G10"
Now that Marlin sends G10 L2(0), pressing 'zero the work offset coordinates button' in CNC.js should result in the work offset coordinate DROs displaying zeros, thanks to fantasmag0ria's changes (**not confirmed, not committed**)

"not really automatic - does it work or nor"
I _too_ think it does work (speculation)

"work coordinate reporting"
Moving away from an ad-hoc format to a structured JSON format seems inevitable.  It would be good if the ad-hoc positional data that is about to be added already resembles proven g2core JSON key/value pairs to encourage a move in that direction.  Transition to JSON can wait if it breaks gcode parsers out there.

**Could I ask you to please post a snippet of a g2core JSON positional report, if you have one handy, thank you.**

Many thanks for your hard work.

MitchBradley, 2020-01-10:
https://github.com/synthetos/g2/wiki/Status-Reports#json-mode-status-reports

JayPerez1, 2020-11-26:
> https://github.com/synthetos/g2/wiki/Status-Reports#json-mode-status-report

> I did not spend much time looking at the changes because they are not presented in the standard form of a pull request (PR) against a specific version. There is a very good reason for that standard [...]

Hello Mitch Bradley, I came across this thread while searching for help on getting the Feedrate Override in CNCjs to work with Marlin. I'm not sure if this is a Marlin or CNCjs issue or where to post a question on it.

Would you please be so kind as to point me to the proper site to find any information on this or to post a question ?

Thanks in advance for any help you can provide.

Jay

MitchBradley, 2020-11-26:
It appears to me that cncjs already has feedrate overrides for Marlin.  There is this UI code in cncjs/src/app/widgets/Marlin/Overrides.jsx:

```
        <div className={styles.overrides}>
            {!!ovF && (
                <DigitalReadout label="F" value={ovF + '%'}>
                    <RepeatButton
                        className="btn btn-default"
                        style={{ padding: 5 }}
                        onClick={() => {
                            controller.command('feedOverride', -10);
                        }}
                    >
                        <i className="fa fa-arrow-down" style={{ fontSize: 14 }} />
                        <span style={{ marginLeft: 5 }}>
                        -10%
                        </span>
                    </RepeatButton>

```
and this server code in cncjs/src/server/controllers/Marlin/MarlinController.js:

```
            // Feed Overrides
            // @param {number} value A percentage value between 10 and 500. A value of zero will reset to 100%.
            'feedOverride': () => {
                const [value] = args;
                let feedOverride = this.runner.state.ovF;

                if (value === 0) {
                    feedOverride = 100;
                } else if ((feedOverride + value) > 500) {
                    feedOverride = 500;
                } else if ((feedOverride + value) < 10) {
                    feedOverride = 10;
                } else {
                    feedOverride += value;
                }
                // M220: Set speed factor override percentage
                this.command('gcode', 'M220S' + feedOverride);
```

supported by this Marlin command:

https://marlinfw.org/docs/gcode/M220.html

JayPerez1, 2020-11-26:
Thank you for the quick reply and the information !

There is a widget for Feedrate Override but it doesn't seem to be working.

I've shutdown the system for the night but will look at the M220 in the
morning to see if I can manually change the feedrate.

Jay

Domush, 2022-02-07:
Hi all,

I wanted to give this thread a poke, since I have added a few features to Marlin which will greatly enhance CNCjs compatibility/usability.

https://github.com/MarlinFirmware/Marlin/pull/23688 - G10 offset support
https://github.com/MarlinFirmware/Marlin/pull/23689 - Work+Machine position reporting

That last one contains _quite a few_ CNC enhancements which can be leveraged by CNCjs, namely brief status reporting:

**Here's a rundown of the new feature set:**
```
/**
 * Send machine status updates to host (Useful for CNC/Laser)
 *
 * - Auto-report state of the machine (like GRBL does) in numerical format.
 * - Auto-report position during long moves.
 *
 * Possible Marlin statuses (GRBL-compatible equivalent)
 * 0: MF_INITIALIZING    (M_INIT)
 * 1: MF_SD_COMPLETE     (M_ALARM)
 * 2: MF_WAITING         (M_IDLE)
 * 3: MF_STOPPED         (M_END)
 * 4: MF_RUNNING         (M_RUNNING)
 * 5: MF_PAUSED          (M_HOLD)
 * 6: MF_KILLED          (M_ERROR)
 */
// #define REPORT_STATUS_TO_HOST // Send regular machine status updates to host in GRBL format ("S_XYZ: #")
#if ENABLED(REPORT_STATUS_TO_HOST)
  // Report status during long moves
  // Only enable if needed, it is very verbose and will cause the console to scroll quickly
  // #define REPORT_STATUS_DURING_MOVES
#endif

// Use an abbreviated, advanced status report designed to relay maximum information to the host
// in the smallest number of bytes (not very human-readable, designed for error-free host UI updates)
// Example: |#|X10.000:Y100.000:Z0.5|X20.000:Y120.000:Z10.5|1|0|1|2|110|2
// Legend:
//  Status header [|#], Work coords (G92..), Machine coords (G53), Metric [1] (G21), Absolute pos. mode [0] (G90),
//  Tool #[1] (active_extruder), Coord set [2] (G55), Feed rate override [110]% (M220 S110), Status: MF_WAITING [2]
//
// As you can see, quite a bit of info is transmitted to hosts which support it, saving multiple verbose queries.
#define COMPACT_STATUS_REPORTS
#if ENABLED(COMPACT_STATUS_REPORTS)
  // M114 will return compact reports instead of the usual verbose report
  // #define M114_USES_COMPACT_REPORTS // CAUTION - May break host UI's which look for M114's verbose output
  // #define GRBL_COMPATIBLE_STATES    // Report GRBL-equivalent states instead of Marlin states
#endif
```

telleropnul, 2022-02-07:
Hi Edward, 

These changes are much appreciated as on the surface this will bring
GRBL features to Marlin.  Have a look here at what the main developer
from CNC.js Mitch Bradley had to say about wpos: tags like in g2core. 

Many thanks for your hard work! 

> Hi all, 
> 
> I wanted to give this thread a poke, since I have added a few features to Marlin which will greatly enhance CNCjs compatibility/usability. 
> 
> MarlinFirmware/Marlin#23688 [1] - G10 offset support
> MarlinFirmware/Marlin#23689 [2] - Work+Machine position reporting 
> 
> That last one contains _quite a few_ CNC enhancements which can be leveraged by CNCjs, namely brief status reporting: 
> 
> HERE'S A RUNDOWN OF THE NEW FEATURE SET: 
> 
> /**
> * Send machine status updates to host (Useful for CNC/Laser)
> *
> * - Auto-report state of the machine (like GRBL does) in numerical format.
> * - Auto-report position during long moves.
> *
> * Possible Marlin statuses (GRBL-compatible equivalent)
> * 0: MF_INITIALIZING    (M_INIT)
> * 1: MF_SD_COMPLETE     (M_ALARM)
> * 2: MF_WAITING         (M_IDLE)
> * 3: MF_STOPPED         (M_END)
> * 4: MF_RUNNING         (M_RUNNING)
> * 5: MF_PAUSED          (M_HOLD)
> * 6: MF_KILLED          (M_ERROR)
> */
> // #define REPORT_STATUS_TO_HOST // Send regular machine status updates to host in GRBL format ("S_XYZ: #")
> #if ENABLED(REPORT_STATUS_TO_HOST)
> // Report status during long moves
> // Only enable if needed, it is very verbose and will cause the console to scroll quickly
> // #define REPORT_STATUS_DURING_MOVES
> #endif
> 
> // Use an abbreviated, advanced status report designed to relay maximum information to the host
> // in the smallest number of bytes (not very human-readable, designed for error-free host UI updates)
> // Example: |#|X10.000:Y100.000:Z0.5|X20.000:Y120.000:Z10.5|1|0|1|2|110|2
> // Legend:
> //  Status header [|#], Work coords (G92..), Machine coords (G53), Metric [1] (G21), Absolute pos. mode [0] (G90),
> //  Tool #[1] (active_extruder), Coord set [2] (G55), Feed rate override [110]% (M220 S110), Status: MF_WAITING [2]
> //
> // As you can see, quite a bit of info is transmitted to hosts which support it, saving multiple verbose queries.
> #define COMPACT_STATUS_REPORTS
> #if ENABLED(COMPACT_STATUS_REPORTS)
> // M114 will return compact reports instead of the usual verbose report
> // #define M114_USES_COMPACT_REPORTS // CAUTION - May break host UI's which look for M114's verbose output
> // #define GRBL_COMPATIBLE_STATES    // Report GRBL-equivalent states instead of Marlin states
> #endif

  

Links:
------
[1] https://github.com/MarlinFirmware/Marlin/pull/23688
[2] https://github.com/MarlinFirmware/Marlin/pull/23689
[3] https://github.com/cncjs/cncjs/issues/549#issuecomment-1031042174
[4]
https://github.com/

MitchBradley, 2022-02-07:
I am not the CNCjs main developer; that is Cheton Wu.  I have contributed very little code to cncjs.