cncjs/cncjs #550: grbl 1.1 Jog command in Axes Widget

JulianS-Uni, 2020-01-07:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.2
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

Is there a way to use the grbl 1.1 Jog commands when using the keyboard navigation in the Axes Widget? When i press and hold an arrow key i can see in the console that CNCjs spams gcode commands to the Arduino. When i stop pressing anything I can see in the console that no more gcode is sended but the axis is still moving. From my understanding it should be possible with grbl jog commands to capture when the button is released and abort the current movement. Isn't this implemented yet?

MitchBradley, 2020-01-07:
cncjs currently does not have support for the new grbl-specific jog commands.  I will add the feature request tag to this issue.

One workaround is to set the jogging increment to a small value so each jog step completes before the keyboard auto-repeats the next one.

JulianS-Uni, 2020-01-07:
Cool, thank you very much!

MitchBradley, 2020-01-08:
This duplicates #512 so I am closing this ticket in favor of the existing one

closed 2020-01-08 by MitchBradley