cncjs/cncjs #555: Can I run cncjs without connecting RPi to router?

karoria, 2020-01-09:
#### Description

I have installed cncjs on RPi4. There are some minor issues which are related to router regarding program execution. The main problem is coordinates and visualizer stops updating for larger programs than 8MB. Also, it takes considerable time while loading new programs. Another problem makes machine pause randomly which can be resumed by start button. I want to run the html locally if possible to get rid of such issues.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop

karoria, 2020-01-09:
I mean network router in above information. ;)

MitchBradley, 2020-01-09:
I'm not entirely clear what you are asking.  If you running the CNCjs UI on the same RPi as the server and the URL begins with localhost, then the data between the UI and the server is probably not going over the outside network.

MitchBradley, 2020-01-09:
Regarding the random pauses - I had a similar problem with one of my milling programs.  It turned out that one of the cuts was just barely exceeding a soft limit, so the controller rejected the move.  Restarting continued the program, but the part had a blemish that was so small that I originally misinterpreted it as tool deflection.  This might not be your problem, but its worth checking, especially if your pauses happen near the machine limits.

karoria, 2020-01-10:
> I'm not entirely clear what you are asking. If you are running the CNCjs UI on the same RPi as the server and the URL begins with localhost, then the data between the UI and the server is probably not going over the outside network.

I use local ip address of RPi like 192.168.0.5:8000 but it seems that the syncing is done through network, not locally. I see different speed of program loading from watch directory with RPi connected to router via ethernet cable and wifi.

karoria, 2020-01-10:
I have confirmed that the random pauses are not due to soft limit as my job is very small and kept at middle of work area. Also it happens inside the job also not just the boundaries.

Jozhua01, 2021-01-27:
I am trying to run cncjs on raspberry pi4. there is no internet in my workshop. Is ti possible to run cncjs?

karoria, 2021-01-29:
> I am trying to run cncjs on raspberry pi4. there is no internet in my workshop. Is ti possible to run cncjs?

Yes. You can. cncjs doesn't need internet. Type http://localhost:8000 in browser (chromium).

LongLiveCHIEF, 2021-02-09:
In addition, the raspberry pi4 can be used as an access point, so you can connect to the raspberry pi directly using your wireless card of your computer.

I do this when I take my 1610 cnc on the road.

https://thepi.io/how-to-use-your-raspberry-pi-as-a-wireless-access-point/

closed 2022-03-27 by emcniece