cncjs/cncjs #557: Support Marlin's advanced ok response

cheton, 2020-01-18:
CNCjs cannot recognizing the advanced ok message as a proper ok and just hangs in a waiting state. Here are some possible combinations according to the Marlin code that responds an advanced ok message:

```
ok P15 B3
ok N10 P15 B3
```

#### Marlin: gcode/queue.cpp

https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/gcode/queue.cpp?fbclid=IwAR1NxIYkG8diSEoc3EEIrn6RNJImfJvdxO7wWWDHLchlub9ZNzMl8RyKdEI

```c
/**
 * Send an "ok" message to the host, indicating
 * that a command was successfully processed.
 *
 * If ADVANCED_OK is enabled also include:
 *   N<int>  Line number of the command, if any
 *   P<int>  Planner space remaining
 *   B<int>  Block queue space remaining
 */
void GCodeQueue::ok_to_send() {
  #if NUM_SERIAL > 1
    const int16_t pn = port[index_r];
    if (pn < 0) return;
    PORT_REDIRECT(pn);
  #endif
  if (!send_ok[index_r]) return;
  SERIAL_ECHOPGM(MSG_OK);
  #if ENABLED(ADVANCED_OK)
    char* p = command_buffer[index_r];
    if (*p == 'N') {
      SERIAL_ECHO(' ');
      SERIAL_ECHO(*p++);
      while (NUMERIC_SIGNED(*p))
        SERIAL_ECHO(*p++);
    }
    SERIAL_ECHOPGM(" P"); SERIAL_ECHO(int(BLOCK_BUFFER_SIZE - planner.movesplanned() - 1));
    SERIAL_ECHOPGM(" B"); SERIAL_ECHO(BUFSIZE - length);
  #endif
  SERIAL_EOL();
}
```

holterb, 2021-04-02:
Wow! so happy I found this, This makes so much sense! I can turn this off for now, but i agree with this feature request!

cheton, 2026-03-13:
It's available in 1.10.9