cncjs/cncjs #561: Text notification upon file completion feature request

vfrman1, 2020-02-19:
#### Description

I'm looking for a pendant that sends a text notification to my phone when a file completes and the CNC goes to an idle state.  This functionality is one of the widgets available on chilipeppr.

#### Versions

- CNCjs: 1.9.2
- Node.js: 6.?
- NPM: 5.?

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x ] Linux

MitchBradley, 2020-02-19:
You can do that with cncjs by setting up an event.  Open the Settings (Gears icon at upper left).  Select Events.  Click on Add+.  Select the Event "G-code: Stop".  Select the Trigger "System".  In the commands box, enter a system command that will send a text.  Search the internet for how to send a text message on your system.  Here is a discussion about how to do in on Raspberry Pi, which probably works on any Linux system: https://www.raspberrypi.org/forums/viewtopic.php?t=69286

vfrman1, 2020-02-19:
> You can do that with cncjs by setting up an event. Open the Settings (Gears icon at upper left). Select Events. Click on Add+. Select the Event "G-code: Stop". Select the Trigger "System". In the [...]

Thanks Mitch!  I didn't even think of setting up an event.

closed 2020-02-19 by vfrman1