cncjs/cncjs #562: Jogging with GRBL

JPluess, 2020-02-19:
i have tried to JOG my CNC instead of using G0 X.... commands it has the benefint that jogging can be stopped 

here is a link. https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging

maybe someone can help me configure it, or if it has not been implemented i would like to request jogging with a stop button. 

here is the stop command: 

0x85 : Jog Cancel

Immediately cancels the current jog state by a feed hold and automatically flushing any remaining jog commands in the buffer.

For now if i send a ! it would stop but then it goes in pause mode. so i must use the 

console interface to send ~ to resume. not very handy

Here is a screenshot, thanks so much

[image]

neilferreri, 2020-03-04:
How are you implementing the $J jog command?  Are you just sending through the console? You cannot cancel a jog unless grbl is in a jog state.

karoria, 2020-03-04:
I have already implemented "$J=" with shopfloor tablet. You need not even need a special button for 0x85 as feedhold can also cancel jog. The best part of it is, when it reaches soft limit, you will not get alarm which needs to be reset. It won't just move and displays a message for the travel exceeded. You can simply jog on other side without reseting and unlocking.

neilferreri, 2020-03-04:
> I have already implemented "$J=" with shopfloor tablet.

I see.  I was hoping you had it implemented in the desktop UI.

karoria, 2020-03-04:
Digging in desktop UI code is too much for me being a non-software person. Somehow I managed to get it work (as well as overrides) on shopfloor tablet. You can try my fork of shopfloor tablet if you wish. You may get some ideas for changing desktop UI accordingly.

JPluess, 2020-03-04:
About the 0x85 command the small 3018 CNC's have no endstops. 😀 Some stop function would be great.

JPluess, 2020-03-04:
Hi there, 
Yes, I entered the values in command line for example: $J=G91X100F500

About the 0x85 command, the small 3018 CNC's have no end stops. 😀 Some stop function would be great. 

For the moment I found -> υ to be working while Jogging.

[image]

neilferreri, 2020-03-05:
@karoria Looking at your modification, I had a question.  Why did you add a G91 before the jog command?  Don't the new jog commands allow you to include a G91 in a non-modal way?

karoria, 2020-03-05:
New jog commands also follow modal state so it is necessary to issue G91 before jogging. I haven't experimented with issuing G91 after "$J=" as it is not written in that way in any example in wiki. I have not taken the modal state back to G90 because in past I had some issues with it in long pressing any jog keys would ignore some state change commands and axis moved rapidly to absolute jog distance location, which were horrible.

neilferreri, 2020-03-13:
If the G91 is applied after the `$J=` it will temporarily change that mode for that jog only. The same is true for G20/G21.

karoria, 2020-03-13:
That's great. I was not knowing about it. Thanks for info.

water-rhatt, 2022-01-01:
I am not understanding this thread. Does CNCjs support $J grlb jogging? I don't understand what "Implementing with shop floor tablet" means. I am using a Linux computer running the latest (as of this post) version of CNCjs, connected to an xPro v5 control box running Grbl 1.3. What would I have to do to get jogging to work? Or is that a feature that is not yet implemented?

Thanks.

MitchBradley, 2022-01-01:
The main CNCjs UI does not support the $J form of jogging.  It only support the old kind where you send a G91 G0 command.

There is an alternative to the main UI at https://github.com/cncjs/cncjs-shopfloor-tablet .  That shopfloor tablet UI has fewer features but many people find it easier to use in some circumstances, particularly for production use when the display is a tablet computer.  It is programmed in plain old Javascript, unlike the main UI which uses very fancy modern Javascript.  Some people find it easier to understand and modify the older dialect because there is much less to learn.  As such, @karoria has successfully modified the shopfloor tablet UI code to use $J jogging.

While the shopfloor tablet UI is intended for tablet use, it will work on a Linux system.  You have to do some configuration work to mount the directory that contains the shopfloor tablet code.  There is information somewhere in the wiki for how to do that, but I don't have time to look it up.