cncjs/cncjs #564: Windows compatible console

JPluess, 2020-02-19:
I am using the latest CNCjS on a raspberry PI. 

the console is nice, but i miss a way to copy paste code i have from a file in windows.

here is an example image of what it could look like: 
[image]

it would be nice to have a dropdown where specific commands can be pinned or unpinned with a dropdown function. (maybe add a config in settings menu to choose what commands are used frequently) 

here is an image example of reprap webinterface by  chrishamm:
[image]

thanks a lot.

MitchBradley, 2020-02-19:
It is already possible to paste into the console window using right-click Paste when the cursor is over the black area of the window:
[image]

Development on CNCjs is rather slow right now, so if you would like to see a lot of changes implemented soon, you or someone else will need to do some JavaScript programming.

JPluess, 2020-03-04:
I have investigated a little. It works in Microsoft Edge, but not in Firefox. Is there a chance the "paste" command can be enabled?

[image]

I understand your situation. I am no programmer so I can’t help.