cncjs/cncjs #566: Custom Grbl startup message confusing CNCjs?

etx, 2020-02-22:
#### Description

I'm trying to run CNCjs with a controller from Millright. It looks like they customized the startup message and I think this is confusing CNCjs. The Jog controls are greyed out no matter what I try. I can give the controller commands through the console fine. Their startup looks like this:

`Grbl 1.1i MegaV 4 Axis Router ['$' for help]`

I tried editing the Regex in GrblLineParserStartup.js to `/^([a-zA-Z0-9]+)\s+((?:\d+\.){1,2}\d+[a-zA-Z0-9\-\.]*)\s+([a-zA-Z0-9]+\s+)?(\[[^\]]+\])/` But it didn't work for me. I'm no Regex ninja.

Just curious what the suggested fix could be. CNCjs looks like exactly what I need, looking forward to using it. 

#### Versions

- CNCjs: 1.10.x
- Node.js: 12.x
- NPM: 6.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [X] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [X] Mac
- [ ] Linux

karoria, 2020-02-22:
Some controllers need "reset" and "unlock" after connection. Try it. I am also using customized startup message, but cncjs runs well.

etx, 2020-02-22:
Hrm, I did try that after searching on the issue. No amount of resetting and unlocking seems to fix it for me.

etx, 2020-02-22:
Noodling it a little more I landed on this Regex:
`^([a-zA-Z0-9]+)\s+((?:\d+\.){1,2}\d+[a-zA-Z0-9\-\.\s+]*)\s+(\[[^\]]+\])`

Now it's working, woohoo! I added \s+ in the second capture group to account for the additional characters after the version number. This sets the additional info in the version property I guess, but haven't noticed a problem yet. 

Example: 
https://regex101.com/r/ff88u1/1

If this is of any use I'll gladly submit a PR. I'd like to suggest people using this new machine use CNCjs instead of UGS.

closed 2020-02-22 by etx

karoria, 2020-02-22:
Thanks for sharing the solution.