cncjs/cncjs #568: A way to test for the existence of variables in a macro

crosma, 2020-02-22:
Currently, to my knowledge, there is no way to test if a variable is set in a macro and trying to use a variable that does not exist can crash CNCjs. For example in the two part tool change/offset macros if you accidentally run the second macro first it will crash CNCjs and you must start over, rehome the machine and whatnot.

I propose a macro command such as

`%test global.state.PROBE_X_LOCATION`

That will test if global.state.PROBE_X_LOCATION is set and if not, end the macro with an error, possibly a dialog.

neilferreri, 2020-06-09:
@crosma You could just set up a macro like this:
`(Test = [global.state.MYVARIABLE])`

@cheton What would really be useful is if we could test that a value is not `undefined` before using it in a macro.

crosma, 2020-08-31:
@neilferreri, the goal would be to stop the macro if the variable is not set. Currently if an unassigned variable is used it crashes CNCjs's back end.