cncjs/cncjs #569: Limit Axes Widget Jog Distances to Custom Only

bcreigh, 2020-02-22:
#### Description

It would be nice to be able to limit the jog distances to only the custom distances that I entered into the Axes settings. I really only want a handful of options for imperial and metric (ex. 10, 1, 0.1, 0.01) and would save me from doing a bunch of scrolling each time I go to change distance.

#### Versions

- CNCjs: 1.9.20
- Node.js: 10.5.2
- NPM: 6.4.1

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

chetmurphy, 2020-02-22:
I would like to see this implemented also. I use 50, 5, 1, .1, .01 …

MitchBradley, 2020-02-22:
It is unlikely that this feature will be implemented soon, so as a workaround, you might try this:

Go to Settings(Gears Icon)>Workspace, Export the existing configuration to a file, edit that .json file with a text editor and put the selections you want inside the metric distances array - "distances": [0.01, 0.1, 1, 10]

That will not restrict the dropdown to only those items, but the new ones will appear at the top of the list (even if they duplicate ones below), so they can be selected easily without scrolling.

bcreigh, 2020-02-24:
thanks Mitch!

I had to delete my previous custom distances and re-enter. They do show up at the top as you said. Yay! This is great. I can live with this :)

In the more recent versions (not sure when implemented) you can enter the custom distances in the settings for the Axes widget. This saves having to export and import settings.