cncjs/cncjs #570: Touchscreen compatible settings

karoria, 2020-02-23:
#### Description

I am using 13 inch touch display, still I can't press a button confidently as the buttons are really small and very nearby. I need suggestions for what can be done to make it touch friedly. Further I need very basic functions of jog, file selection, grbl status, overrides, etc. Can someone give me a clue on what changes can make UI more touch friendly. I have also considered shopfloor tablet but I am missing override control and alarm description there.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

hoffma0901, 2020-03-29:
Have you tried to use the shopfloor-tablet extension? Even on a 7" Display it is easy to use.

emcniece, 2022-03-27:
Can we close this in favour of https://github.com/cncjs/cncjs-shopfloor-tablet/issues/18 ?