cncjs/cncjs #571: Visualizer hangs while loading larger files

karoria, 2020-02-23:
#### Description

I have noticed that file doesn't load if it is larger, say 20MB. Making 3D view button OFF doesn't help. I have also noticed that even if you make 3d view off, it renders the toolpath in the background as I can see the rendering cubical icon in the graphics area at that time. So, for now it is ok if a feature can be developed where user can limit file size for toolpath rendering somewhere in settings. When it exceeds this limit, it should not attempt graphical rendering and should show the gcode lines instead.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

nickm324, 2020-09-27:
Is there any update on this? I loaded a 18mb gcode file and the UI crashes every  time and will not update any more. If I refresh the browser it will just try and render the 3D view again and the UI crashes again. I cannot find a way to disable the 3D View completely so that i just have the scrolling Gcode view which is fine.

karoria, 2020-09-28:
I am also waiting. @cheton is there a workaround so that we can disable 3d in code so that it doesn't try to load the 3d toolpath? GUI button of making 3d off doesn't actually help as mentioned above.