cncjs/cncjs #578: Unable to configure custom widget

kebinshorts, 2020-03-12:
#### Description

When provided the URL to a GitHub repository, the custom widget interface becomes blank. I have tried this with the "ReactApp" custom widget that came with CNCjs. In addition, I have tried publishing the repo to GitHub pages and using that URL instead, which similarly does not work. I have tried this with both NPM CNCjs and the CNCjs electron app. Am I doing something wrong?

Link to repo: https://github.com/kebinshorts/ReactApp
[image]
[image]

#### Versions

- CNCjs: 1.9.21
- Node.js: 6.17.1
- NPM: 6.14.2

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [x] Mac
- [ ] Linux

cheton, 2020-03-12:
https://kebinshorts.github.io/ReactApp

kebinshorts, 2020-03-12:
The url you provided shows "no serial connection" even when the longmill is plugged in.
[image]

cheton, 2020-03-13:
Ooops. I forgot that the path to a custom widget must be specified within the mount points.

Here is my configuration:
```js
{
    "state": {
        "checkForUpdates": true,
        "controller": {
            "exception": {
                "ignoreErrors": false
            }
        }
    },
    "mountPoints": [
        {
            "route": "/widget",
            "target": "https://cncjs.github.io/cncjs-widget-boilerplate/v1/"
        }
    ]
}
```

Also see https://github.com/cncjs/cncjs#file-format for details.

After that, set the URL field to `/widget`, you will see the content on the custom widget:

[image]

closed 2020-03-29 by cheton