cncjs/cncjs #579: Ability to let firmware handle M0

Tailslide, 2020-03-15:
Whenever I get a tool change I have to confirm the M0 both in my firmware LCD screen and on cnc.js.  I'd love a configuration setting to have cnc.js not pause on M0 and let the firmware handle notifications so I don't have to confirm them twice.. at the machine and on cnc.js

jeffeb3, 2020-04-06:
Or, if cnc.js is handling it, then maybe it should not send it to the LCD. I can see users wanting either option.

dmagyar, 2021-08-13:
I second this - it's so annoying to run back to my cncjs after I click the button on the LCD. I think CNCJS should wait until an "ok" comes back from Marlin indicating that the user has confirmed the pause.

MitchBradley, 2021-08-15:
You might be surprised how tricky it can be to track the machine state accurately enough to synchronize the M0 correctly.  I solved the problem for g2core in cncjs-shopfloor-tablet but it was not easy.  Making it work correctly across all controllers was so complicated that I quit working on it. No way to get compensated for the time it was taking.  The fundamental problem is that there are lots of queues between the UI and the physical machine, and the controller firmwares are not particularly good at reporting instantaneous state.  CNCjs can detect M0 and stop sending, but it is tricky to determine when the controller has processed the M0 - which is not the same as when it has received and acknowledged the command.  In the meantime, there can be other kinds of hold conditions arising from things like GCode errors, tool changes, feedholds sent either from the UI or from a button, etc.  None of the controller serial protocols are very well designed; g2core's is the best of the bunch but it too has problems.

MitchBradley, 2021-08-15:
I see that the original request was for CNCjs to ignore M0.  That would be easy to do - just comment out the line
```
                        this.workflow.pause({ data: 'M0' });
```
in src/server/MarlinController.js .  If you wait for an official release to incorporate the change, you might be waiting a long time, considering how little cncjs development has occurred in the last couple of years.
That said, it would certainly be possible for CNCjs to ignore M0 and just pass it through.