cncjs/cncjs #581: jog not working on marlin

ameisso, 2020-03-21:
#### Description

We are trying to use cncjs with a on a Marlin/miniRambo cnc controller. 
So far everything is working great (sending gcodes, controlling the movement, spindle...) 
We are encountering problems with the axes control widget, when moving the axis with gcode, the position is updated on cncjs accordingly. But so far, pressing the buttons X+, Y+... that are not greyed nothing happens. Like if no code is sent to the controller. 

Is their an option to activate somewhere or something we are missing ? Or is this a real issue ? 

PS : We've previously used cncjs with a Smoothie board and the jog is working ok. 

Best 

#### Versions

- CNCjs: 1.9.21 / app version 
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [X] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core
- [X]Marlin
#### Hardware

- [] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [x] Mac
- [ ] Linux

MitchBradley, 2020-03-21:
When you  press the jog buttons, do you see anything in the console window?  Here's what I see:

```
Cap:CHAMBER_TEMPERATURE:0
feeder> G91
ok
feeder> G0 X1
ok
feeder> G90
ok
feeder> G91
ok
feeder> G0 X-1
ok
feeder> G90
ok
>
```
This is with Marlin 2.0 running on an ESP32, connected via USB serial to a PC running the CNCjs app under Windows.  The

djinn5150, 2020-03-22:
I am having same problem, no output on the console when pressing buttons in the widget, however.  If I send say g0 x0 in the console or really anything, then everything i've pressed plays out in the console and the machine follows the code, then back to a hang.  I am running marlin 2.0 on an skr 1.3 connected via usb

ameisso, 2020-03-22:
Hello all, 

We're having the same issue as @djinn5150, typing G0 X1 in the console moves the motors, but hitting the buttons of the jog don't moves anything (witch is normal because nothing is sent in the console)

djinn5150, 2020-03-22:
FWIW, This same build of Marlin was working fine with octoprint, its got to be something in the way cncjs/marlin communicates.  I tried turning on #advanced_ok and #notimeouts neither seemed to help.  At initial connection the console would display the machine capabilities report followed by T:0 then hang.  If I pressed a series of buttons in the Axis widget it would queue them up until I typed something in the console to get a response from marlin, once that response came the feeder would play out in the console then go back to a hanging state.  It was almost as if cnc.js was looking for a specific response it wasn't getting.  For me it at this time it just made more sense to switch from Marlin to smoothie for my situation. Smoothieware is working just fine so far.  @ameisso  if your ultimate goal is use of cnc.js maybe Grbl-mega would be a better solution then marlin, unless you need marlin for other reasons of course.

ameisso, 2020-03-23:
Thanks everyone for the inputs. 
This confirms our feeling that there is something wrong with cncjs and Marlin at the moment. 

Is their something else to do to reports issues ? 
I will try to sort it out once the confinement ends here. 

All the best

jaredwolff, 2020-03-23:
Confirming this here too.

jaredwolff, 2020-03-23:
It appears that the commands are emitted from the UI:

```
2020-03-23T19:57:20.119Z - debug service:cncengine socket.command("/dev/tty.usbmodem14401", "gcode"): id=ik1yXqDlQQcXam-cAAAB     at Socket.<anonymous> (/Users/jaredwolff/.nvm/versions/node/v10.15.3/lib/node_modules/cncjs/dist/cncjs/server/index.js:18275:25)
2020-03-23T19:57:20.120Z - debug service:cncengine socket.command("/dev/tty.usbmodem14401", "gcode"): id=ik1yXqDlQQcXam-cAAAB     at Socket.<anonymous> (/Users/jaredwolff/.nvm/versions/node/v10.15.3/lib/node_modules/cncjs/dist/cncjs/server/index.js:18275:25)
2020-03-23T19:57:20.120Z - debug service:cncengine socket.command("/dev/tty.usbmodem14401", "gcode"): id=ik1yXqDlQQcXam-cAAAB     at Socket.<anonymous> (/Users/jaredwolff/.nvm/versions/node/v10.15.3/lib/node_modules/cncjs/dist/cncjs/server/index.js:18275:25)
```

That would be the `G91`, the `G0 XYZ` and then the `G90`

But one of these conditions is preventing it from being sent:

```
                { // The following criteria must be met to trigger the feeder
                    const notBusy = !(this.history.writeSource);
                    const senderIdle = (this.sender.state.sent === this.sender.state.received);
                    const feederIdle = !(this.feeder.isPending());

                    if (notBusy && senderIdle && feederIdle) {
                        this.feeder.next();
                    }
                }
```
(https://github.com/cncjs/cncjs/blob/3250c1b23180fc6e7947238381c76e7d36a5284f/src/server/controllers/Marlin/MarlinController.js#L1273)

MitchBradley, 2020-03-23:
Best guess: Something about the form of Marlin's response is confusing CNCjs into thinking that a command has not been acknowledged.  This is one of the problems with Marlin - it is so highly configurable that no two builds will behave quite the same.

MitchBradley, 2020-03-23:
from 
From src/server/controllers/Marlin/MarlinLineParserResultOk.js:

```
class MarlinLineParserResultOk {
    // ok
    static parse(line) {
        const r = line.match(/^ok$/);
```

Anything other than a bare "ok" will not be recognized as acknowledging previous lines.

MitchBradley, 2020-03-23:
So far nobody except me has shown data from their console window.  Hint, hint.

jaredwolff, 2020-03-23:
Hey @MitchBradley 

Here's mine:

```
CNCjs 1.9.21 [Marlin]
Connected to /dev/tty.usbmodem14401 with a baud rate of 250000
start
echo:Marlin 2.0.5.1
echo: Last Updated: 2020-03-16 | Author: (none, default config)
echo:Compiled: Mar 23 2020
echo: Free Memory: 5462  PlannerBufferBytes: 1152
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm (mm)
echo:; Steps per unit:
echo: M92 X1517.88 Y1517.88 Z1517.88 E500.00
echo:; Maximum feedrates (units/s):
echo:  M203 X7.00 Y7.00 Z7.00 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X50.00 Y50.00 Z50.00 E10000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P50.00 R50.00 T50.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Stepper driver current:
echo:  M906 X1200 Y1200 Z1200
echo:  M906 T0 E800
echo:; Hybrid Threshold:
echo:  M913 X6 Y6 Z6
echo:  M913 T0 E30
echo:; StallGuard threshold:
echo:  M914 X9 Y9 Z2
echo:; Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 T0 E
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
FIRMWARE_NAME:Marlin 2.0.5.1 (GitHub) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:0 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:0
Cap:VOLUMETRIC:0
Cap:AUTOREPORT_TEMP:0
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:CHAMBER_TEMPERATURE:0
ok T:0
> g
echo:Unknown command: "g"
ok
feeder> G91
ok
feeder> G0 X-1
ok
feeder> G90
ok
ok T:0
```

As you alluded to, there is the output from the `M105` command. (`ok T:0`) Is that mucking things up?

jaredwolff, 2020-03-23:
Looks like it has to do with not having a temperature probe/extruder. Changing M105.cpp to get rid of the `T:0` seems to have done it for me in the Marlin code. Not sure if there's a way to address it in cnc.js (or if it's worth it). 

```
From 21eb59071fd76c46a13753a811e11091c05d1927 Mon Sep 17 00:00:00 2001
From: Jared Wolff <hello@jaredwolff.com>
Date: Mon, 23 Mar 2020 16:32:56 -0400
Subject: [PATCH] Only way to get jogging working

---
 Marlin/src/gcode/temp/M105.cpp | 10 +++-------
 1 file changed, 3 insertions(+), 7 deletions(-)

diff --git a/Marlin/src/gcode/temp/M105.cpp b/Marlin/src/gcode/temp/M105.cpp
index 7d68ccadc..5c99912a6 100644
--- a/Marlin/src/gcode/temp/M105.cpp
+++ b/Marlin/src/gcode/temp/M105.cpp
@@ -40,12 +40,8 @@ void GcodeSuite::M105() {
         , parser.boolval('R')
       #endif
     );
-
-    SERIAL_EOL();
-
-  #else
-
-    SERIAL_ECHOLNPGM(" T:0"); // Some hosts send M105 to test the serial connection
-
+  
   #endif
+
+  SERIAL_EOL();
 }
-- 
2.23.0
```

MitchBradley, 2020-03-23:
T:0 is the result of M105 if HAS_TEMP_SENSOR is false.  HAS_TEMP_SENSOR is #define HAS_TEMP_SENSOR (HAS_TEMP_HOTEND || BED_OR_CHAMBER || HAS_TEMP_PROBE)

You could try configuring some sort of temperature sensor and see what happens.  Or perhaps change the T:0 in M105.cpp in the Marlin source to " T:293.0 /0.0 B:25.9 /0.0 @:0 B@:0".  If that fixes it, we need to improve the regular expression in marlin line parser result temperature.

MitchBradley, 2020-03-23:
Ideas crossed in the mail :-)

MitchBradley, 2020-03-23:
Here's something you can try:
Go into index.js - on my Windows box, using the app, it is at /Users/wmb/AppData/Local/Programs/cncjs-app/resources/app/server/index.js .  Search for "T:293".  Just below that you will see a line that begins "var r = line.match(".  Replace it with:

`      var r = line.match(/^(?:ok)?(?:\s*(T|B|T\d+):([0-9\.\-]+)(?:\s+\/([0-9\.\-]+)(?:\s+\((?:[0-9\.\-]+)\))?)?)*(?:\s*(@|B@|@\d+):([0-9\.\-]+))*(?:(W):(\?|[0-9]+))?/i);`

Then exit and restart CNCjs.  That worked for me, with versions of Marlin that had full and abbreviated temperature reports.

MitchBradley, 2020-03-23:
I used https://www.debuggex.com/?flavor=javascript

This is the regex in the current cncjs master: 
`    ^(?:ok)?\s*(?:(?:(T|B|T\d+):([0-9\.\-]+)\s+\/([0-9\.\-]+)(?:\s+\((?:[0-9\.\-]+)\))?)|(?:(@|B@|@\d+):([0-9\.\-]+))|(?:(W):(\?|[0-9]+)))
`
With this visualization:

[image]

[Debuggex Demo](https://www.debuggex.com/r/Jm8j8OcxXZnIQ-Vo)

It fails to recognize not only the T:0 form, but also many of the other forms that are shown in comments in MarlinLineParserResultTemperature.js

This is the proposed new regex (different from the one suggested in my previous message in this thread):
`    ^(?:ok)?(?:\s*(T|B|T\d+):([0-9\.\-]+)(?:\s+\/([0-9\.\-]+)(?:\s+\((?:[0-9\.\-]+)\))?)?)+(?:\s*(@|B@|@\d+):([0-9\.\-]+))*(?:\s*(W):(\?|[0-9]+))?$`

[image]

[Debuggex Demo](https://www.debuggex.com/r/STkYNdiQ9naUJDfG)

It successfully recognizes all the forms shown in the commentary, and T:0, and rejects many corrupted versions of the forms.

jaredwolff, 2020-03-24:
@MitchBradley awesome! I'll test it on my end and let you know if I run into any hiccups with my limited setup (no temp sensors).

ameisso, 2020-03-24:
Hello Everyone, 

First of all thanks for the support ! 
We were able to fix the problem by modifying the Marlin firmware 

in M105.cpp replacing : 
`SERIAL_ECHOLNPGM(" T:0"); // Some hosts send M105 to test the serial connection `
by 
` SERIAL_ECHOLNPGM(" T:293.0 /0.0 B:25.9 /0.0 @:0 B@:0");`

does the trick 

We will wait for the next electron release to test the cncjs modification. 

All the best

closed 2020-03-29 by cheton

cheton, 2020-03-29:
This issue was resolved in [1.9.22](https://github.com/cncjs/cncjs/releases/tag/v1.9.22)

MitchBradley, 2023-06-28:
Pretty much everything in Marlin can be disabled with a compile option.

Fusseldieb, 2023-06-29:
> Pretty much everything in Marlin can be disabled with a compile option.

Yes, but how/where!? I know that I could strip that temperature report away if I fiddle with the cop files, but it isn't a feasible way. Also, CNCjs should be able to understand a simple temperature report.

MitchBradley, 2023-06-29:
Maybe you should ask on a Marlin forum.  We are not experts on Marlin.  The vast majority of CNCjs users use GRBL so dealing with every Marlin thing is not the highest priority for the very limited amount of ongoing CNCjs development.  Marlin is a fast-moving target - there were 6 commits to the code base in the last 7 days, not counting commits that just affect the build/release system. That is about the number of feature commits that CNCjs had in the year 2022.  What I am implying is that you are much more likely to get a timely resolution of your problem by finding how to suppress the message in the Marlin source, compared to waiting for someone in the CNCjs world to take an interest in this problem that affects very few users.

cheton, 2023-06-29:
It appears that the heater state has changed over the years, and it may be necessary to update CNCjs to support new tokens such as `C`, `P`, `L`, `M`, `S`, and `R`.

https://github.com/MarlinFirmware/Marlin/blob/aee8e15fd2b1b701a5f16e92a0c557c6dccb13de/Marlin/src/module/temperature.cpp#L4140-L4242

```cpp
  /**
   * Print a single heater state in the form:
   *        Bed: " B:nnn.nn /nnn.nn"
   *    Chamber: " C:nnn.nn /nnn.nn"
   *      Probe: " P:nnn.nn"
   *     Cooler: " L:nnn.nn /nnn.nn"
   *      Board: " M:nnn.nn"
   *        SoC: " S:nnn.nn"
   *  Redundant: " R:nnn.nn /nnn.nn"
   *   Extruder: " T0:nnn.nn /nnn.nn"
   *   With ADC: " T0:nnn.nn /nnn.nn (nnn.nn)"
   */
  static void print_heater_state(const heater_id_t e, const_celsius_float_t c, const_celsius_float_t t
    OPTARG(SHOW_TEMP_ADC_VALUES, const float r)
  ) {
    char k;
    bool show_t = true;
    switch (e) {
      default:
        #if HAS_TEMP_HOTEND
          k = 'T'; break;
        #endif
      #if HAS_TEMP_BED
        case H_BED: k = 'B'; break;
      #endif
      #if HAS_TEMP_CHAMBER
        case H_CHAMBER: k = 'C'; break;
      #endif
      #if HAS_TEMP_PROBE
        case H_PROBE: k = 'P'; show_t = false; break;
      #endif
      #if HAS_TEMP_COOLER
        case H_COOLER: k = 'L'; break;
      #endif
      #if HAS_TEMP_BOARD
        case H_BOARD: k = 'M'; show_t = false; break;
      #endif
      #if HAS_TEMP_SOC
        case H_SOC: k = 'S'; show_t = false; break;
      #endif
      #if HAS_TEMP_REDUNDANT
        case H_REDUNDANT: k = 'R'; break;
      #endif
    }
    #define SFP _MIN(SERIAL_FLOAT_PRECISION, 2)

    SString<50> s(' ', k);
    if (TERN0(HAS_MULTI_HOTEND, e >= 0)) s += char('0' + e);
    s += TS(':', p_float_t(c, SFP));
    if (show_t) { s += F(" /"); s += p_float_t(t, SFP); }
    #if ENABLED(SHOW_TEMP_ADC_VALUES)
      // Temperature MAX SPI boards do not have an OVERSAMPLENR defined
      s.append(F(" ("), TERN(HAS_MAXTC_LIBRARIES, k == 'T', false) ? r : r * RECIPROCAL(OVERSAMPLENR), ')');
    #endif
    s.echo();
    delay(2);
  }

  void Temperature::print_heater_states(const int8_t target_extruder
    OPTARG(HAS_TEMP_REDUNDANT, const bool include_r/*=false*/)
  ) {
    #if HAS_TEMP_HOTEND
      print_heater_state(H_NONE, degHotend(target_extruder), degTargetHotend(target_extruder) OPTARG(SHOW_TEMP_ADC_VALUES, rawHotendTemp(target_extruder)));
    #endif
    #if HAS_HEATED_BED
      print_heater_state(H_BED, degBed(), degTargetBed() OPTARG(SHOW_TEMP_ADC_VALUES, rawBedTemp()));
    #endif
    #if HAS_TEMP_CHAMBER
      print_heater_state(H_CHAMBER, degChamber(), TERN0(HAS_HEATED_CHAMBER, degTargetChamber()) OPTARG(SHOW_TEMP_ADC_VALUES, rawChamberTemp()));
    #endif
    #if HAS_TEMP_COOLER
      print_heater_state(H_COOLER, degCooler(), TERN0(HAS_COOLER, degTargetCooler()) OPTARG(SHOW_TEMP_ADC_VALUES, rawCoolerTemp()));
    #endif
    #if HAS_TEMP_PROBE
      print_heater_state(H_PROBE, degProbe(), 0 OPTARG(SHOW_TEMP_ADC_VALUES, rawProbeTemp()));
    #endif
    #if HAS_TEMP_BOARD
      print_heater_state(H_BOARD, degBoard(), 0 OPTARG(SHOW_TEMP_ADC_VALUES, rawBoardTemp()));
    #endif
    #if HAS_TEMP_SOC
      print_heater_state(H_SOC, degSoc(), 0 OPTARG(SHOW_TEMP_ADC_VALUES, rawSocTemp()));
    #endif
    #if HAS_TEMP_REDUNDANT
      if (include_r) print_heater_state(H_REDUNDANT, degRedundant(), degRedundantTarget() OPTARG(SHOW_TEMP_ADC_VALUES, rawRedundantTemp()));
    #endif
    #if HAS_MULTI_HOTEND
      HOTEND_LOOP() print_heater_state((heater_id_t)e, degHotend(e), degTargetHotend(e) OPTARG(SHOW_TEMP_ADC_VALUES, rawHotendTemp(e)));
    #endif
    SString<100> s(F(" @:"), getHeaterPower((heater_id_t)target_extruder));
    #if HAS_HEATED_BED
      s.append(" B@:", getHeaterPower(H_BED));
    #endif
    #if HAS_HEATED_CHAMBER
      s.append(" C@:", getHeaterPower(H_CHAMBER));
    #endif
    #if HAS_COOLER
      s.append(" C@:", getHeaterPower(H_COOLER));
    #endif
    #if HAS_MULTI_HOTEND
      HOTEND_LOOP() s.append(F(" @"), e, ':', getHeaterPower((heater_id_t)e));
    #endif
    s.echo();
  }
```

cheton, 2023-06-29:
> Experiencing this issue with the latest CNCjs. It queues commands up until I type something in the console. Then, it runs everything at once and gets stuck again. It seemingly happened after I enabled a COOLER temp sensor, as it reports the temp to the host, and there's no option to disable it in Marlin (afaik).
> 
> ```
> CNCjs 1.10.2 [Marlin]
> Conectado à porta /dev/ttyACM0 com taxa de transferência de 115200
> > .
> echo:Unknown command: "."
> ok
> ok L:21.84 /0.00 @:0 C@:0
> >
> ```
> 
> It looks like it doesn't acknowledge the "ok L:21.84 /0.00 @:0 C@:0" and hangs. Any jog command after that doesn't get executed anymore.

A typical reply supported by CNCjs would usually include the `T` parameter. For examples:

```
ok T:0
ok T:293.0 /0.0 B:25.9 /0.0 @:0 B@:0
ok T:293.0 /0.0 B:25.9 /0.0 T0:293.0 /0.0 T1:100.0 /0.0 @:0 B@:0 @0:0 @1:0
ok T:293.0 /0.0 (0.0) B:25.9 /0.0 T0:293.0 /0.0 (0.0) T1:100.0 /0.0 (0.0) @:0 B@:0 @0:0 @1:0
ok T:293.0 /0.0 (0.0) B:25.9 /0.0 T0:293.0 /0.0 (0.0) T1:100.0 /0.0 (0.0) @:0 B@:0 @0:0 @1:0 W:?
ok T:293.0 /0.0 (0.0) B:25.9 /0.0 T0:293.0 /0.0 (0.0) T1:100.0 /0.0 (0.0) @:0 B@:0 @0:0 @1:0 W:0
  T:293.0 /0.0 B:25.9 /0.0 @:0 B@:0
  T:293.0 /0.0 B:25.9 /0.0 T0:293.0 /0.0 T1:100.0 /0.0 @:0 B@:0 @0:0 @1:0
  T:293.0 /0.0 (0.0) B:25.9 /0.0 T0:293.0 /0.0 (0.0) T1:100.0 /0.0 (0.0) @:0 B@:0 @0:0 @1:0
```

However, based on the provided examples, it seems that the `T` parameter is missing from the report. It appears that the `HAS_TEMP_HOTEND` option has been disabled, causing CNCjs to be unable to recognize the reply.

Fusseldieb, 2023-06-29:
Thanks for looking into the issue @cheton .

> Does your Marlin firmware serve for a purpose other than being a 3D printer?

It is actually a CO2 Laser. The temperature probe that's connected is to monitor the water temperature internally (and shut off the board if it gets past a threshold). As soon as I enabled it, it broke CNCjs.

cheton, 2023-06-30:
@Fusseldieb 

I created a PR (#832) that allow parsing temperaure data like heated chamber, cooler, and other temperature states, specifically for Marlin 2.x.

However, I don't have a board with customized Marlin 2.x firmware, I would like to inquire about the platform on which you run the CNCjs app for further verification needs.

Fusseldieb, 2023-06-30:
> @Fusseldieb
>
> I created a PR (#832) that allow parsing temperaure data like heated chamber, cooler, and other temperature states, specifically for Marlin 2.x.
>
> However, I don't have a board with [...]

Wow! I saw your PR and it looks good :)

> I would like to inquire about the platform on which you run the CNCjs app for further verification needs.
I run CNCjs on a Raspberry Pi 3 with PM2.

cheton, 2023-07-01:
@Fusseldieb 

[v1.10.3](https://github.com/cncjs/cncjs/releases/tag/v1.10.3), has been released. Feel free to check if this release can resolve your issue.

Fusseldieb, 2023-07-03:
> @Fusseldieb
> 
> [v1.10.3](https://github.com/cncjs/cncjs/releases/tag/v1.10.3) has been released. Feel free to check if this release can resolve your issue.

Just updated and tried homing and jogging. Working as expected :)

Thanks for your time!

hm2dev, 2024-04-07:
I just tried marlin with 1.10.3 and jogging, as well as sending gcode and most other things did not work.

I found this bug and the MarlinLineParserResultOk hint stated above by @Fusseldieb.

> from From src/server/controllers/Marlin/MarlinLineParserResultOk.js:
> 
> ```
> class MarlinLineParserResultOk {
>     // ok
>     static parse(line) {
>         const r = line.match(/^ok$/);
> ```
> 
> Anything other than a bare "ok" will not be recognized as acknowledging previous lines.

I changed the regular expression to not care about anything comming after the ok.

With the changes to cncjs-app\resources\app\server\controllers\Marlin\MarlinLineParserResultOk.js
Line 21: 
   function parse(line) {
      var r = line.match(/^ok/);
      if (!r) {

Everything is working now. 

So I would suggest this problem still exists in 1.10.3.

cheton, 2024-10-02:
@hm2dev Could you please assist in collecting the logs for further analysis?

commit f33e6464 (2020-03-29): Fix #581 - jog not working on Marlin (#583)

* Fix #581 - jog not working on Marlin

This RE works better for parsing Marlin temperature reports:
* It handles the "ok T:0" case that Marlin issues when it has no temperature probes
* It allows B@ and W fields to be optionally present after T:, T0:, and T1: blocks, instead of as alternatives to them.
* It recognizes all the test cases shown in the code commentary, whereas the original fails on most of them.
https://www.debuggex.com/?flavor=javascript is a really great way to visualize and test regular expressions!

See the trail in issue #581 for [... cut]
commit 931c2e83 (2020-03-29): Fix #581 - jog not working on Marlin (#583)

* Fix #581 - jog not working on Marlin

This RE works better for parsing Marlin temperature reports:
* It handles the "ok T:0" case that Marlin issues when it has no temperature probes
* It allows B@ and W fields to be optionally present after T:, T0:, and T1: blocks, instead of as alternatives to them.
* It recognizes all the test cases shown in the code commentary, whereas the original fails on most of them.
https://www.debuggex.com/?flavor=javascript is a really great way to visualize and test regular expressions!

See the trail in issue #581 for [... cut]