cncjs/cncjs #582: eslint-loader Multiple exports error on Mac OSX build

jaredwolff, 2020-03-24:
#### Description

When attempting to do an `nom install` on Mac OSX  10.14.6, it gets all the way to `cross-env NODE_ENV=development webpack-cli --progress --config webpack.config.server.development.js && npm run build-i18n-server` and then fails at the `eslint-loader` module.

Here's the error output from `npm install`:

```
> cncjs@1.9.21 build-dev-server /Users/jaredwolff/CNC/cncjs
> cross-env NODE_ENV=development webpack-cli --progress --config webpack.config.server.development.js && npm run build-i18n-server

Hash: be26aabb4374491e884a                                                        
Version: webpack 4.30.0
Time: 14404ms
Built at: 03/23/2020 4:12:27 PM
   Asset      Size  Chunks             Chunk Names
index.js  1.61 MiB   index  [emitted]  index
Entrypoint index = index.js
[0] multi ./index.js 28 bytes {index} [built]
[./app.js] 17 KiB {index} [built]
[./config/settings.js] 459 bytes {index} [built]
[./index.js] 11.6 KiB {index} [built]
[./lib/ensure-type.js] 903 bytes {index} [built]
[./lib/logger.js] 3.26 KiB {index} [built]
[./lib/urljoin.js] 420 bytes {index} [built]
[./services/cncengine/index.js] 264 bytes {index} [built]
[./services/configstore/index.js] 8.32 KiB {index} [built]
[./services/monitor/index.js] 1.58 KiB {index} [built]
[bcrypt-nodejs] external "bcrypt-nodejs" 42 bytes {index} [built]
[chalk] external "chalk" 42 bytes {index} [built]
[dns] external "dns" 42 bytes {index} [built]
[ensure-array] external "ensure-array" 42 bytes {index} [built]
[expand-tilde] external "expand-tilde" 42 bytes {index} [built]
    + 180 hidden modules

ERROR in ./api/index.js
Module Error (from /Users/jaredwolff/CNC/cncjs/node_modules/eslint-loader/index.js):

/Users/jaredwolff/CNC/cncjs/src/server/api/index.js
   2:1  error  Multiple exports of name 'get'       import/export
   3:1  error  Multiple exports of name 'fetch'     import/export
   4:1  error  Multiple exports of name 'get'       import/export
   6:1  error  Multiple exports of name 'fetch'     import/export
   6:1  error  Multiple exports of name '__delete'  import/export
   6:1  error  Multiple exports of name 'update'    import/export
   6:1  error  Multiple exports of name 'create'    import/export
   6:1  error  Multiple exports of name 'read'      import/export
   7:1  error  Multiple exports of name 'fetch'     import/export
   7:1  error  Multiple exports of name '__delete'  import/export
   7:1  error  Multiple exports of name 'create'    import/export
   7:1  error  Multiple exports of name 'read'      import/export
   7:1  error  Multiple exports of name 'update'    import/export
   8:1  error  Multiple exports of name 'update'    import/export
   8:1  error  Multiple exports of name 'create'    import/export
   8:1  error  Multiple exports of name 'read'      import/export
   8:1  error  Multiple exports of name '__delete'  import/export
   8:1  error  Multiple exports of name 'fetch'     import/export
   9:1  error  Multiple exports of name 'create'    import/export
   9:1  error  Multiple exports of name 'read'      import/export
   9:1  error  Multiple exports of name '__delete'  import/export
   9:1  error  Multiple exports of name 'update'    import/export
   9:1  error  Multiple exports of name 'fetch'     import/export
  10:1  error  Multiple exports of name 'read'      import/export
  10:1  error  Multiple exports of name '__delete'  import/export
  10:1  error  Multiple exports of name 'update'    import/export
  10:1  error  Multiple exports of name 'create'    import/export
  10:1  error  Multiple exports of name 'fetch'     import/export
  11:1  error  Multiple exports of name 'update'    import/export
  11:1  error  Multiple exports of name 'create'    import/export
  11:1  error  Multiple exports of name 'fetch'     import/export
  11:1  error  Multiple exports of name 'read'      import/export
  11:1  error  Multiple exports of name '__delete'  import/export

✖ 33 problems (33 errors, 0 warnings)

 @ ./app.js 41:0-29 308:52-61 313:59-70 315:50-59 316:51-60 317:56-65 319:50-59 320:51-60 321:59-68 322:60-69 325:56-71 327:53-65 328:54-66 329:57-69 330:57-69 331:63-75 332:62-74 334:51-61 335:53-63 336:55-65 337:55-65 338:61-71 340:53-65 341:54-66 342:57-69 343:57-69 344:63-75 346:51-61 347:52-62 348:55-65 349:55-65 350:61-71 352:48-55 353:49-56 354:48-55 355:52-59 356:52-59 357:58-65 359:50-59 360:52-61 361:54-63 362:54-63 363:60-69 365:56-65 366:57-66 367:55-64 368:56-65
 @ ./index.js
 @ multi ./index.js
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! cncjs@1.9.21 build-dev-server: `cross-env NODE_ENV=development webpack-cli --progress --config webpack.config.server.development.js && npm run build-i18n-server`
npm ERR! Exit status 2
npm ERR! 
npm ERR! Failed at the cncjs@1.9.21 build-dev-server script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/jaredwolff/.npm/_logs/2020-03-23T20_12_27_511Z-debug.log
```

npm log attached.

[2020-03-23T20_12_27_511Z-debug.log](https://github.com/cncjs/cncjs/files/4372460/2020-03-23T20_12_27_511Z-debug.log)

**Note:** I am running `npm install` in the root of the repository.

#### Versions

- CNCjs: 1.9.21
- Node.js: v10.19.0
- NPM: 6.14.3

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core
- [x] Marlin

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [x] Mac
- [ ] Linux

MitchBradley, 2020-03-24:
@cheton will need to look at this, but in the short term, you might be able to make progress by going into package.json and removing the clause "&& npm run eslint" from the "prepare:" line.  Ditto for "stylint" and "test" if they cause problems too.  I haven't tested this...

cheton, 2020-03-29:
The issue was fixed in the master branch.

closed 2020-03-29 by cheton