cncjs/cncjs #584: [GRBL] Pausing to idle after every completed movement

aososam, 2020-04-02:
#### Description

My GRBL 0.8.2 Machine does the homing cycle without a problem but during a job it goes to Idle after some movements, I have to press pause and then play again so it continues with the program, but it is really difficult to complete a job pausing and resuming every three lines of code.
I tried with another sender (Candle) and it sends the codes without pausing or going to idle at any moment. (but I like CNC.js better I just wish it did not went to idle during the job). If anyone has a similar issue or if you know how to fix it please let me know. if there is no fix I may be able to contribute to fixing it with some instructions. Thank you a lot I am attaching my code.
[program.txt](https://github.com/cncjs/cncjs/files/4420769/program.txt)

#### Versions

- CNCjs: 1.9.22
- Node.js: 13.6.0
- NPM: 6.13.14

#### How Do You Install CNCjs?

- [ ] NPM
- [X] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [X] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [X] Mac
- [ ] Linux

ameisso, 2020-04-02:
I am experiencing the same issue (with grbl 0.8) 
Will try to update to the last version and see how it goes

aososam, 2020-04-02:
Ok Please let me know if it worked. Thank you.

ameisso, 2020-04-02:
I've just updated to the last version of grbl and it works far better now. 
All the best

closed 2020-04-02 by MitchBradley

aososam, 2020-04-03:
I had some problems after Updating Grbl, the axis X and Z were not moving and the X was moving only to the positive direction. Everything was solved following this guide 
https://www.laseraxe-customize.com/forum/laser-engraving-machine/update-grbl-1-1f