cncjs/cncjs #586: serialport: Is a directory

AlexandrParkhomenko, 2020-04-03:
yay  cncjs
1 aur/cncjs 1.9.20-1 (+0 0.00%) 
    Web-based interface for CNC controllers running Grbl, Marlin, Smoothieware, or TinyG.
==> Packages to install (eg: 1 2 3, 1-3 or ^4)
==> 1
:: Checking for conflicts...
:: Checking for inner conflicts...
[Aur: 1]  cncjs-1.9.20-1

:: Downloaded PKGBUILD (1/1): cncjs
  1 cncjs                                    (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 
:: Parsing SRCINFO (1/1): cncjs
==> Making package: cncjs 1.9.20-1 (Fri 03 Apr 2020 11:34:19 AM MSK)
==> Retrieving sources...
  -> Downloading v1.9.20.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   120  100   120    0     0    351      0 --:--:-- --:--:-- --:--:--   351
100 12.4M    0 12.4M    0     0  2521k      0 --:--:--  0:00:05 --:--:-- 3046k
  -> Found cncjs
  -> Found cncjs.service
  -> Found cncjs.sysusers
==> Validating source files with sha512sums...
    v1.9.20.tar.gz ... Passed
    cncjs ... Passed
    cncjs.service ... Passed
    cncjs.sysusers ... Passed
==> Making package: cncjs 1.9.20-1 (Fri 03 Apr 2020 11:34:24 AM MSK)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found v1.9.20.tar.gz
  -> Found cncjs
  -> Found cncjs.service
  -> Found cncjs.sysusers
==> Validating source files with sha512sums...
    v1.9.20.tar.gz ... Passed
    cncjs ... Passed
    cncjs.service ... Passed
    cncjs.sysusers ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Extracting v1.9.20.tar.gz with bsdtar
==> Starting prepare()...
==> Sources are ready.
==> Making package: cncjs 1.9.20-1 (Fri 03 Apr 2020 11:34:26 AM MSK)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Starting build()...
yarn install v1.22.4
info No lockfile found.
[1/5] Validating package.json...
[2/5] Resolving packages...
warning @trendmicro/react-datepicker > react-datepicker > react-popper > popper.js@1.16.1: You can find the new Popper v2 at @popperjs/core, this package is dedicated to the legacy v1
warning @trendmicro/react-datepicker > prop-types > fbjs > core-js@1.2.7: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
warning @trendmicro/react-popover > rc-align > babel-runtime > core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
warning babel-polyfill > core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
warning bcrypt-nodejs@0.0.3: bcrypt-nodejs is no longer actively maintained. Please use bcrypt or bcryptjs. See https://github.com/kelektiv/node.bcrypt.js/wiki/bcrypt-vs-brypt.js to learn more about these two options
warning hogan.js > mkdirp@0.3.0: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
warning jimp > mkdirp@0.5.1: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
warning jimp > request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
warning opencollective > babel-polyfill > core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
warning react-bootstrap > @babel/runtime-corejs2 > core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
warning react-dropzone > attr-accept > core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
warning react-router-redux@5.0.0-alpha.9: This project is no longer maintained.
warning babel-cli > babel-register > core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
[... 308 lines cut ...]
      |                         ^~~~
/home/alexandr/.cache/node-gyp/13.12.0/include/node/node.h:618:36: note: in definition of macro ‘NODE_MODULE_X’
  618 |       (node::addon_register_func) (regfunc),                          \
      |                                    ^~~~~~~
../src/serialport.cpp:485:1: note: in expansion of macro ‘NODE_MODULE’
  485 | NODE_MODULE(serialport, init);
      | ^~~~~~~~~~~
make: *** [serialport.target.mk:110: Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory '/p/AUR/cncjs/src/cncjs-1.9.20/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Linux 5.5.13-arch1-1
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /p/AUR/cncjs/src/cncjs-1.9.20/node_modules/serialport

==> ERROR: A failure occurred in build().
    Aborting...
Error making: cncjs
1 2m56s 11:37:09 

cncjs/src/cncjs-1.9.20/node_modules/serialport
-bash: cncjs/src/cncjs-1.9.20/node_modules/serialport: Is a directory

cheton, 2020-04-10:
Try downgrading your Node.js from 13.12.x to 10.x to install CNCjs 1.9.2x.

AlexandrParkhomenko, 2020-04-21:
I'll try docker image

closed 2024-10-17 by cheton