cncjs/cncjs #589: Operation randomly stopping (logs: Stop sending G-code: hold=true)

Billiam, 2020-04-12:
#### Description

I'm having an issue wint my jobs seemingly halting at random. I'm not seeing errors in the console, or the error log, but when using verbose logging I can see this in the output.

```
debug controller:Grbl Stop sending G-code: hold=true, sent=51, received=51
```

I'm not sure what to make of this, or what it's indicating, or how to fix it.

No errors related to the serial port, USB cable is very short. This is occurring with the router and stepper motors off entirely as well.

#### Versions

- CNCjs: 1.9.22
- Node.js: 10.19.0
- NPM: 6.14.4

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi 3B+
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

Billiam, 2020-04-12:
After much testing, this is only occurring when using the shopfloor-tablet pendant. Moving report to https://github.com/cncjs/cncjs-shopfloor-tablet/issues/20

closed 2020-04-12 by Billiam

MitchBradley, 2020-04-12:
One way to get more information: Open Developer Tools in your browser and look at the console log. In cncjs-shopfloor-tablet/src/app.js there is this code:
```
controller.on('serialport:read', function(data) {
    console.log('sr', data);
```
That generates a fair amount of console logging information.
The code in the CNCjs server that generates the "Stop sending G-code" message only triggers if the workflow state is "Paused", and there are only a few conditions that cause a workflow pause, namely:
* M0 or M1 pause
* M6 tool change
* Receiving an error report from GRBL
* An explicit "gcode:pause" message sent from the UI
The error from GRBL is the one I suspect at the moment.  That will show up in the console log as a "sr" message containing an error report as the data.