cncjs/cncjs #590: 3D view rendering upside down on Raspberry Pi

AnthonyMyatt, 2020-04-13:
#### Description

The 3D view is rendering upside down (tool is pointing upwards)

#### Versions

- CNCjs: 1.9.22
- Node.js: 12.16.2
- NPM: 6.14.4

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [] Grbl
- [ ] Smoothieware
- [x ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

[image]
[image]

MitchBradley, 2020-04-13:
Have you tried with different gcode files, including some very simple ones? If it soes not happen with simple files, try to find out which aspect of the file that triggers the problem is necessary to make it fail

AnthonyMyatt, 2020-04-13:
Hello @MitchBradley,
Thank you for your reply. Unfortunately this issue is not file specific; as you can see from the second photo this happens with no file loaded. The second photo shows the "top" view with no file loaded, in my case the numbers are flipped (as "top" is actually viewing from the bottom, so you can see the back of the numbers).

maroussel, 2020-04-13:
@AnthonyMyatt I have a similar setup on a RPI4 + RPI touch screen on G2Core and I just went to look because I remembered I seen something I found strange on the axis marking... But unfotunately I went and loaded my only 3 gcode files that I have and they are all fine. I'll let you know if I'm able to reproduce.

AnthonyMyatt, 2020-04-14:
@maroussel thanks for looking into this.
I'm curious how the RPI4 is performing? My RPI3 is sluggish when running a job and the UI starts to lock up. From what I understand the RPI4 is supposed to be a massive leap forward in performance, trying to work out if I should spend the $110AUD on a RPI4 or use an old Intel Atom media centre PC I have lying around.

maroussel, 2020-04-14:
@AnthonyMyatt the RPI4 perform without any issue so far. I use the full UI with 3D rendering and I didn't noticed any lag. I would say for now it perform the same as my laptop. I want to test it with more complex designs and maybe streaming a webcam. I liked the fact that the RPI4 was easily able to run fanless and was fast to boot. (I use a heatsink case) Since you already have the Atom media centre PC it's an harder choice.

AnthonyMyatt, 2020-04-16:
Thanks @maroussel, I think I will probably go with the RPI4 in the long run.

I have identified that the problem is with WebGL in general and not specific to cncjs, any WebGL content renders upside down on my RPI3.

AnthonyMyatt, 2020-04-16:
I fixed the issue by turning on "Override software rendering list" under chrome://flags. Apparently on the RPI3 GPU acceleration is forced off in Chromium and turning this option on ignores that and uses GPU acceleration anyway. I guess there is a bug with WebGL and software rendering.

@maroussel I guess Chromium supports GPU acceleration on the RPI4.

closed 2020-04-16 by AnthonyMyatt

maroussel, 2020-04-16:
It make sense! You are right, mostlikey chromium supports GPU acceleration on the RPI4/Buster. I didn't really got in on the RPI2 because it was way too slow for rendering anyway. I'm glad you did find the cause, it was not an obvious config.