cncjs/cncjs #591: Webhook when M5 or M6 runs

carlos-sarmiento, 2020-04-18:
#### Description

I'd like to receive a web notification when an m5 or m6 command is executed. I'd like to get my home automation system to notify me either I should shut down the spindle, or there is a tool change pending.
 
#### Versions

- CNCjs: 1.9.22
- Node.js: 10
- NPM: (unknown)

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2020-04-18:
Settings (gear icon upper left) > Events > Add > Trigger: GCode: Pause , System command to run a script that talks to your home automation.

closed 2020-04-18 by MitchBradley

carlos-sarmiento, 2020-04-19:
@MitchBradley tried what you suggested but the M6 does not trigger a pause event. Execution stops and I get a chance to run a macro, but the GCode: Pause trigger isn't run

carlos-sarmiento, 2020-04-19:
I do get a GCode: Resume event once I click on the play button to continue the file

carlos-sarmiento, 2020-04-19:
So I checked the code and "gcode:pause" is only triggered by Visualization.jsx when you click the pause button.