cncjs/cncjs #593: [RFC] Add specific events for M5 and M6 commands

carlos-sarmiento, 2020-04-19:
#### Description

I've started using CNCjs recently and would like to have a mechanism to run arbitrary commands when an M5 or M6 command is executed. Right now that functionality is not there (the controller files don't have any code checking for an M5 and a M6 doesn't trigger any event).

I've submitted #592 to add the already existing 'gcode:pause' event to M0, M1 and M6 commands. Although M0 and M1 are actual pauses, M6 is not conceptually one so I feel like it should be its own (separate) event. 

My intention is to get aligned with the maintainers of the project if they would like to see such a contribution before I start writing the code. 

#### Versions

- CNCjs: 1.9.22
- Node.js: 10
- NPM: unknown

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

mmdoogie, 2020-06-30:
I've got a personal branch where I implemented basically this, if you want to take a look at how I tackled it.  I only implemented it for Grbl but the other controllers should be very similar.

Haven't submitted this as a PR either because I've got a couple small ones that I haven't got feedback on yet as well.

(mmdoogie/cncjs@more-gcode-events)[/mmdoogie/cncjs/tree/more-gcode-events]

TJ-Houghton, 2021-02-07:
Additionally, for M6, is it possible to display a dialog box that display the Tool Number that the program is expecting?

closed 2026-03-13 by cheton