cncjs/cncjs #597: Enable keyboard jogging whenever machine is idle

skendrot, 2020-04-29:
#### Description
Enabling keyboard jogging after every run gets very frustrating. Especially since you have to use the mouse to enable keyboard jogging. There should be a setting to always enable keyboard jogging whenever grbl is idle.

#### Versions

- CNCjs: 1.9.20

#### How Do You Install CNCjs?

- [x] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

jeremysugnaux, 2020-05-03:
Would be so useful! Or maybe a keyboard shortcut to enable keyboard jogging?

sorynl, 2024-06-29:
yes please, this would be such a useful functionality