cncjs/cncjs #601: Is there the ability to send G-Code right into CNCjs like OctoPrint does via Kiri:Moto

dlarue, 2020-05-11:
#### Description

I'm looking for an way to send gcode files to CNCjs from Kiri:Moto like it can send files to OctoPrint.
Currently I export from Kiri:Moto slicer and then upload into CNCjs.  Having the ability to send the file right into CNCjs or even just into the monitored directory would be very handy.

#### Versions

- CNCjs: 1.9.22
- Node.js: 10.20.1
- NPM: 6.14.4

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

dlarue, 2020-05-15:
FWIW, I just learned that when I save the gcode from Kiri:Moto the download tab at the bottom of Chrome can be dragged and dropped onto CNCjs.  So this feature of OctoPrint-like streaming to the CNCjs server process isn't high priority for me to use these to packages together in my workflow.

stewartoallen, 2020-05-17:
I would like to see this as well. Given that Kiri:Moto already has multiple ways to send a file via http(s), establishing an endpoint on the cncjs http(s) server that sets CORS headers to receive a cross-domain POST should be possible.

mmdoogie, 2020-06-21:
There is an HTTP API already that you can do that with.  You just have to pass the parameters port, name, and gcode as form fields.  The only tricky part is either having a username/password set up for authentication or generating the JSON Web Token from the .cncrc secret.

For example, the following works with a valid token inserted:
`curl -F "port=/dev/ttyWhatever" -F "name=code.nc" -F "gcode=<code.nc" -H "Authorization: Bearer token-string-here" "http://cncjs:8000/api/gcode"`

stewartoallen, 2020-07-07:
Kiri:Moto will not be able to post to this endpoint because of web security rules. Kiri:Moto is served from an https url. Your local cnc.js instance is not. In order for this to work, the cnc.js instance would need to also run on an https port, set CORS headers, be setup in DNS, and have valid certificates that match the DNS name.

I just made an OctoPrint Plugin available that works around these issues by bouncing print jobs off of a secure endpoint on grid.space -- this solves the above issues and also allows for zero configuration discovery of local printer targets. The same could be done for cnc.js. Since I do not think it has a plugin system, it would have to be incorporated into the core cnc.js code.

I am thinking of making a separate npm package that is just a listener with a file-drop target. This would be a standalone process that performs the same function as the GridBot controller and OctoPrint plugin and provide a seamless way to spool gcode directly to the directory cnc.js is watching.