cncjs/cncjs #608: Android App?

ecornbower, 2020-06-05:
Has anyone tried converting this to an Android app? It would be great to be able to use a cheap Android tablet to control your CNC.

Looks like it might be pretty straightforward with something like [AndroidJS](https://android-js.github.io/).

MitchBradley, 2020-06-05:
CNCjs has a server component that connects to the controller (GRBL for example) via a USB serial connection. Most tablets do not have an easy way to make that connection, although it might be possible to use a Bluetooth serial adapter or some form of USB OTG adapter.

I use an Android tablet as the control surface by running the CNCjs server component on an inexpensive Raspberry Pi Zero W that talks over WiFi to a $100 10" Android tablet.  The tablet is not connected by wires to the machine, so it is not subject to the many forms of electrical interference that can cause problems in CNC setups. I can move it anywhere that is convenient without having wires tethering it to the machine.

I wrote the cncjs-shopfloor-tablet UI specifically for this setup, because the normal CNCjs UI is difficult to use on a touchscreen - the controls are too small to touch reliably.

ecornbower, 2020-06-05:
@MitchBradley Thanks for the input. There are existing Android apps that can connect to a CNC via USB serial, but none as nice as CNC JS.

I'd ultimately like to use a Bluetooth connection for the reasons you cite. If no one else has tried to create an Android server/front-end app combo, I'll probably start working on it.

MitchBradley, 2020-06-05:
I hope that you join the CNCjs user group and provide support for your app, because I am going crazy trying to support all the different CNCjs platform combinations.

ecornbower, 2020-06-05:
Ha yeah I bet, I'd be happy to provide support if it works out.