cncjs/cncjs #613: disable 3D view still results in high processing usage

dlarue, 2020-06-18:
#### Description

There should be a way to disable 3D rendering and processing when not needed or desired.

Running on Chrome Linux with 3D View disabled I can feel the GPU fan running and heat dissipation.

MitchBradley, 2020-06-18:
Have you tried Chrome command line options?  That's all that I found in an internet search.  I'm not sure there is a way for code running on the browser to directly control that.  If you know of a way, please tell us.

dlarue, 2020-06-18:
I have not but will look into what command line options are available.

I've built up a ESP32 on a DB25 adapter board running Grbl_Esp32 and because of the 3.3V signalling have been chasing noise and this has lead me to running CNCjs for hours over the past few days.  This is really heated up the left side of the laptop and while turning 3D Viewer off helps it doesn't seem to eliminate the high usage of that tab.  I'll shutdown the browser and look into the commandline options. Thanks.

MitchBradley, 2020-06-18:
The Radeon GPU on my desktop machine has a control panel with all sorts of controls of various sorts.  I think there might be a way to disable GPU usage for certain apps either there or in some OS control panel.  Regardless, it is probably not in CNCjs's purview.

closed 2020-06-18 by MitchBradley

dlarue, 2020-06-18:
So I shutdown CNCjs with 3D View turned off and shutdown the browser and restarted it.

Without CNCjs loaded I see 0% and 5% on the GPU indicator.

With CNCjs running but no g-code loaded I don't see any impact much at all

With CNCjs running and g-code loaded AND starting with 3D view disabled then I see 80-90% GPU utilization... 

80-90% GPU utilization for scrolling g-code text?

dlarue, 2020-06-18:
So with 3D View enabled but turning off Tool Animation GPU usage goes down to 10-15% so that's good.  But I was quite surprised to see when I then switched off the 3D View, GPU usage bounced back to 85-90%.  That seems very odd.

dlarue, 2020-06-20:
hopefully some of this helps with display lagging the machines so much.