cncjs/cncjs #616: MACRO - User input prompt

rgodin974, 2020-06-24:
Is it possible into the macro widget to have an user input prompt ? It could be very useful to increase the macro’s possibilities. The functionality is in bCNC. Thank you.

neilferreri, 2020-06-28:
That would be cool!

aolshove, 2020-06-30:
Yes please!

mmdoogie, 2020-06-30:
@aolshove PR #615 would let messages like that be done at least, let me know what you think of that implementation for the messages?  There's some screenshots over there.

aolshove, 2020-07-01:
Yes I think it might.  My only concern is that the M0 throws up a dialog that may block the display of the message?  It seems like a parameter included in the M0 command for a "message" might be a better solution.  Just a though.  I would use the new MSG gcode command when it's implemented though.

mmdoogie, 2020-07-02:
@aolshove agree, it would be cool to see both.  Should not get covered up though, except on maybe a really small screen.

[image]