cncjs/cncjs #624: G-code gets messed up

JohanBoshoff74, 2020-07-09:
#### Description

When I design a project via Easel and then convert it to g-code and load on CNCjs there seems to be a data scramble or misinterpretation. For a week my CNCjs was working perfectly with Easel g-code. Then suddenly yesterday the code seems to have gone bad. On the original design the cutting depth is set to 3mm. But when I load it into CNCjs the cutting suddenly goes down to about 5mm. I have not changed any settings on CNCjs since I started using it a week ago, so I can't think where the problem might come in. Any advice please.

#### Versions

- CNCjs: 1.9.x

#### How Do You Install CNCjs?

- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl

#### Hardware

- [ ] Desktop or Laptop

#### Operating System

- [ ] Windows

neilferreri, 2020-07-09:
Can you share your Easel project and gcode?

JohanBoshoff74, 2020-07-10:
How do I do that? Do I copy the page address and post it here?

jkesseler, 2020-07-16:
> How do I do that? Do I copy the page address and post it here?

In Easel go to `File > Share` and select `Share with link` and post the link in this thread. Here is an [example of a project of mine](https://easel.inventables.com/projects/oAXaF5OrMuTlyPKD8BF4UA)
Alternatively you can post the generated G-Code on [pastebin](https://pastebin.com/) and leave the link in this thread.