cncjs/cncjs #627: Desktop app on Windows 7 doesn't remember Maximized state

Muralha, 2020-07-16:
#### Description

Every time I open the Desktop app on Windows 7, the window is always close to full size but it never remembers the Maximize state I set before (the maximize button on the top right corner).
I always have to maximize (with precision) so I don't click outside the window by mistake.
Would it be possible for the desktop app (at least on Windows) to remember the maximized state?
[image]

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

GeoffreyOlivier, 2021-09-01:
Hi, where you found your app desktop please ?

Muralha, 2021-09-01:
> Hi, where you found your app desktop please ?

[https://github.com/cncjs/cncjs/releases](url)