cncjs/cncjs #629: Single Axis Homing in GRBL

Cows2Computers, 2020-07-21:
#### Description

The buttons to home a single axis issues a command of G28.2 Y0. G28.2 is not supported in grbl. Can we map those buttons to $HX, $HY, $HZ, which from what I can tell has more support in different flavors of grbl? 

#### Versions

- CNCjs: 1.9.22

#### CNC Controller

- [x] Grbl (specifically grbl_esp32, https://github.com/bdring/Grbl_Esp32)

#### Hardware

- [x] Raspberry Pi

#### Operating System

- [x] Not Applicable

MichMich, 2020-10-12:
Support for this would be great. :)

AustinSaintAubin, 2020-10-12:
Have you thought about creating your own Macro for homing per axis. Doing this would give you a button in the UI to home per axis. This should solve your problem. More info here: https://github.com/cncjs/cncjs/wiki/User-Guide#macro-widget

Cows2Computers, 2020-10-13:
> Have you thought about creating your own Macro for homing per axis. Doing this would give you a button in the UI to home per axis. This should solve your problem. More info here: https://github.com/cncjs/cncjs/wiki/User-Guide#macro-widget

I have done this and it works, but I still have the UI clutter of broken buttons.

MichMich, 2020-10-17:
It would be great if we could overwrite the homing command defaults in the config. If it's not set, it would use the current settings. It this something a PR would be welcome for?

sns5400, 2020-11-14:
Hello,
I have the same problem with the G28.2 home X...Y...Z...A...B...C...also the button on the left for X..Y..Z..A..B..C.. have the same problem it send G28.3 X0 to set the workspace at X0 if you make this editable in the navigation widget settings per axis the solution is universal.

ausip, 2021-12-04:
I'd like to +1 this feature. I am using a xPro V5 controller (based on GRBL_ESP32) and having the same issue. It would be great to be able to customize these standard buttons to support the controller language

cotepat, 2022-02-16:
Yes... does not work either with FluidNC (new version of GRBL for ESP32. Which is awesome by the way)

dwery, 2022-02-16:
While GRBL is not really improved upon anymore, any derivative could very reasonably implement G28.2

Patronics, 2025-09-06:
I have the same issue, (as does the author of issue #654). It seems that GRBL does not support G28.2, but does support the more common `G28` command (without the `.0` on its own, which has effectively the same functionality, and I'd argue is more likely what most people want when clicking the "home" button. Why is there no option to configure what those buttons do? Or at least a more widely supported default?