cncjs/cncjs #630: API Documentation Strategy

timothy-godfrey, 2020-07-22:
#### Description

Hello! I am using cncjs as the foundation for a custom CNC machine interface, and so I need to come to grips with the various APIs in the project. In order to preserve a reference for later use I thought it would be worthwhile to contribute some API documentation. At the moment I am 

- documenting the code with JSDoc comments 
- generating markdown output with documentation.js 
- planning to contribute a page to the cncjs.org project

Does that sound like a good plan? 

It has crossed my mind that an issue on the cncjs.org repository would be the place to discuss contributions to that repository, but I thought it would be worthwhile checking here first, **how would the maintainer/s of cncjs feel about JSDoc comments in the codebase?** Would they be welcome? Is there a preferred strategy?

Thanks!

MitchBradley, 2020-07-22:
Any documentation that you could contribute would be most welcome!

Instead of cncjs.org, I would suggest a github wiki page, as that wiki is more actively maintained than cncjs.org.

@cheton will have to weigh in on the question about JSDoc in the codebase.

timothy-godfrey, 2020-07-23:
Okay. I'll start by contributing the generated markdown into the wiki directly. I'm still pondering build infrastructure for the documentation, but that will probably end up in another branch and a later pull request.

timothy-godfrey, 2020-08-11:
I've decided that writing the  documentation by hand will make more sense than using JSDoc comments in the codebase.This resolves the questions about documentation in the codebase and build infrastructure, so I'll close the issue.

closed 2020-08-11 by timothy-godfrey