cncjs/cncjs #631: Accessing API Endpoints from Boilerplate Widget code

Davies-Cavies, 2020-07-27:
#### Description

I am looking to create a widget that I can mount to CNCjs that can directly load GCode files to CNCjs, as well as start and stop the program and access the parameters of the job like remaining time etc... I have seen that there are API's that can do this, and have found the endpoints within the code but am struggling to implement them into my widget. I am using this widget to communicate to an external source, and so for this problem I assume that I have the gcode file ready within the widget ready to be uploaded.

I am quite new to Javascript programming so I apologise if this seems like a simple matter. Any help will be very much appreciated.

#### Versions

- CNCjs: 1.9.22
- Node.js: 10.21.0
- NPM: 6.14.5

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [x] Mac
- [ ] Linux

timothy-godfrey, 2020-07-28:
Hi! I'm working on a similar goal, and to that end I've been generating documentation in the cncjs project. 

My strategy has been to start by copying functionality from the 'WorkflowControl' widget, so I've been working my way through that. It is the little group of buttons in the top left of the visualizer widget. When events are triggered on the WorkflowControl widget (including file load and upload events) it calls functions from the Visualizer widget (defined in index.jsx in src/widgets/Visualizer passed to WorkflowControl in its 'actions property). The Visualizer has an instance of the class 'Controller' , the bulk of which is defined in a separate repository 'cncjs-controller'. The Visualizer widget uses Controller to send events (with data) to the server. The list of events is defined in cncjs-controller/src/controller.js. 

I'm still working out the relationship (if any) between the Controller class and the other specific controller classes (GrblController, MarlinController etc). I'm also still working through the details of the Visualizer API interface. It does have a function called 'loadGCode' defined in its actions, which appears to load the 3d view of the gcode. This function is called to handle the controller event 'gcode:load'. 

So my rough summary is 
- load gcode directly to the server, using the 'command' function with 'gcode:load' as the event and the gcode as one of the function parameters
- the server will do whatever it does with the gcode and then emit a 'gcode:load' event
- the Visualizer will receive the event (along with whatever else is also listening for it) and render the received gcode.

It's pretty rough, but that's where I'm up to. I'll add more as I go.

timothy-godfrey, 2020-07-28:
Just a caveat, as I read further I wonder if the Visualizer is doing things an old way. There is also cncjs/src/app/api/index.js. Still working my way through it, but I wonder if it is a new 'standardized' API way of handling sending things to the server instead of using a Controller instance. Just a thought; I'm still making my way through it.

Davies-Cavies, 2020-07-30:
Hi Timothy, thank you for the reply. Are you able to direct me towards the documentation that you have written or provide some code examples? I also noticed the API endpoints located there but haven't worked out a way to call these

timothy-godfrey, 2020-08-03:
I am working on the documentation at [API Documentation](https://github.com/cncjs/cncjs/wiki/API-Documentation-(Under-Construction)) It is in early stages so far, and reflects my earlier explorations of WorkflowControl widget. I haven't contributed any documentation to the wiki yet about the server API. I'll be contributing more content over the coming weeks.

I don't have any code examples. Your best bet for examples will be the codebase itself, for the time being. As you have probably found yourself, the cncjs server API is defined in the files found in the directory cncjs/src/server/api. The cncjs codebase uses a client library for that server API, defined in cncjs/src/app/api/index.js. To find code examples of using the API one might search for references to the functions defined in the client library. For example cncjs/app/lib/user.js imports api from 'app/api' and then calls api.signin at line 7. Another example is found in cncjs/src/app/widgets/Axes/index.js at line 595: `res = await api.mdi.fetch();` You can find files that use the client library by grepping for 'app/api' in cncjs/src. (I use atom for development, so I just use the global search).

timothy-godfrey, 2020-08-12:
I've added some more API documentation content to the wiki, and restructured it a little bit:
- [Controller API](https://github.com/cncjs/cncjs/wiki/Controller-API)
- [PubSub API](https://github.com/cncjs/cncjs/wiki/PubSub-API)
- [Server API](https://github.com/cncjs/cncjs/wiki/Server-API)

What strategy are you taking to serve your widget? Are you developing and serving it separately from cncjs and integrating it through the custom widget, or are you adding your widget to the cncjs codebase? Have you decided whether you are going to load your gcode to the server using HTTP (the server API) or websockets (the controller API)?

nickm324, 2020-08-13:
None of this is showing up in the table of contents on the wiki page, are there links to these topics somewhere else?

Davies-Cavies, 2020-08-13:
@timothy-godfrey I have actually managed to access these API's using Axios to create my post and get requests. Took a bit of time to understand what the API endpoints were requiring but then it becomes quite simple after that. For the controller API, do you mean sending the controller.command() as I have achieved this as well by creating an instance of the controller within the widget that can then run these commands like start and stop GCode when required

Davies-Cavies, 2020-08-13:
@timothy-godfrey would be happy to contribute to your documentation with what I have done as it may help others in future that are trying to do similar things

timothy-godfrey, 2020-08-14:
@nickm324 If you just want to read the docs then the links in my earlier post will work. But yes, the navigation in the wiki is a bit hidden. The sidebar 'table of contents' is edited manually (I think) and may not reflect the current state of the wiki. If you click on 'Pages', which appears above the table of contents, then you will see the full list of wiki pages in alphabetic order (and in case you missed it, there's a button at the bottom to expand the list even further). The search filter at the top of the pages list is pretty handy.

@Davies-Cavies My contributions to the documentation are pretty high-level, and lack details on the functions/endpoints. If you want to contribute detailed function documentation, or code examples, I'm sure others will find it helpful in the future.

karoria, 2020-08-30:
Does this mean that a client Web-UI can be made with API only and that doesn't need "cncjs-controller" library to be used? I also want to make a UI for my machine which should be "light weight". Can you suggest best way to do it? Can I send direct socket.io commands to server without using any library? I am not a software person but your little hint can give me a push. BTW, I am good at html, css and JS (beginner).
Thank you in advance. Waiting for your answer.

Davies-Cavies, 2020-08-31:
@karoria What sort of information are you wanting to display on your lightweight UI? The way that we have handled the problem is mounting a widget (look at the documentation for CNCjs widget boilerplate) to the version of CNCjs that is running the machine. This allows the creation of a controller instance which can directly interface with the cncjs program. The widget is written in React (combination of html and Javascript). You can use Socket.io to communicate with that widget from your UI or server depending on what you are wanting to do

karoria, 2020-09-01:
It is a full featured pendant with my choice of layout. But I am wondering, if a UI can communicate over socket.io, why all pendants till date use controller library? BTW, my server and client part both will run on a RPi4.
So basically I am confused with 3 things and can't clearly distinguish all of them:
1. API
2. cncjs controller library for pendants
3. Communicating with socket.io

Davies-Cavies, 2020-09-09:
@karoria what API's specifically are you trying to access? For example trying to upload GCode to that API I used the Axios Node.js module to generate my Post Requests. You can find all of the api code within the server code of the Cncjs program that shows you what parameters are needed for the post. I also believe that you could use the controller command to upload the gcode so long as it is in the right format. 

The controller library has been good for doing things like starting and stopping the program, as mentioned before it is good to make an instance of the controller which is demonstrated many times in the CNCjs code and then add the necessary controller events which is also demonstrated in the code. I have found that analysing the visualiser widget has been good as this deals with starting and stopping the program along with uploading gcode.

karoria, 2020-09-09:
@Davies-Cavies 
I want to make a full featured UI (Pendant), so almost all the necessary commands shall be covered. I just want to make a right choice out of API/socket.io/controller library. To make the code minimum. One of my concerns is program loading for larger programs, say 20MB, which at present doesn't load with main UI or shopfloor UI.
I am using Raspberry Pi 4 (8GB) (same device for server and client) with grbl.