cncjs/cncjs #634: z axis only goes up

thiaggo, 2020-08-17:
#### Description

i cant make the z axis to go down ( it was working fine but something happined)
Z+ goes up
Z- goes up
changed the stepper motor 
changed the stepper driver
changed to x axis slot
only z axis that i cant make it go down

im using grbl 

- CNCjs: 1.9.22
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [ x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [ x] Desktop or Laptop
- [ ] Mobile Device

win 10 64x Operating System

- [] Not Applicable
- [x ] Windows
- [ ] Mac
- [ ] Linux

neilferreri, 2020-08-17:
Check your wiring. Sounds like a bad connection or a broken wire.

thiaggo, 2020-08-18:
> Check your wiring. Sounds like a bad connection or a broken wire.

i think it cant be, i put z axis on x axis and worked fine

MitchBradley, 2020-08-18:
Test it by using the console window to send direct GCode move commands like

G0 Z-1

G0 Z1

It is very unlikely that the problem is in cncjs.  It is more likely to be a problem in the Grbl controller.

Your statements "changed to x axis slot, only z axis that I can make it go down", and "I put z axis on x axis" are impossible to interpret precisely.  Do you mean that you connected the stepper that is physically connected to the mechanical z axis to the X axis driver?  Do you mean that you moved the  driver and the connected stepper?  Since we know very little about your setup, - what kinds of drivers (external, pololu modules, what driver chips), brand of controller board - it is impossible to guess what is going wrong.  The only think I can guess is that it is probably not a problem in CNCjs, so this is not a good place to file an issue.