cncjs/cncjs #638: Can not build:win-x64 on win10

hotpoor, 2020-08-25:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.22
- Node.js: 10.12.0
- NPM: 6.4.1
#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x ] Windows
- [ ] Mac
- [ ] Linux

```
MSI-PC@MSI MINGW64 /d/Xialiwei/github/cncjs (master)
$ npm run build:win-x64

> cncjs@1.9.22 build:win-x64 D:\Xialiwei\github\cncjs
> bash -c 'scripts/electron-builder.sh --win --x64'

/d/Xialiwei/github/cncjs/dist/cncjs /d/Xialiwei/github/cncjs
Cleaning up "/d/Xialiwei/github/cncjs/dist/cncjs/node_modules"
Installing packages...
scripts/electron-builder.sh: line 24: sudo: command not found
scripts/electron-builder.sh: line 25: sudo: command not found
/d/Xialiwei/github/cncjs
===Rebuild native modules using electron
v10.0.0
scripts/electron-builder.sh: line 29: sudo: command not found

> cncjs@1.9.22 electron-builder D:\Xialiwei\github\cncjs
> electron-builder "--win" "--x64"

  鈥?electron-builder  version=22.8.0 os=10.0.18363
  鈥?loaded configuration  file=package.json ("build" field)
  鈥?cancelled by SIGINTon dependencies  platform=win32 arch=x64 appDir=D:\Xialiwei\github\cncjs\dist\cncjs

MSI-PC@MSI MINGW64 /d/Xialiwei/github/cncjs (master)
$   猕?D:\Applications\nodejs\node.exe exited with code ERR_ELECTRON_BUILDER_CANNOT_EXECUTE
Output:

Output:

> @serialport/bindings@2.0.8 install D:\Xialiwei\github\cncjs\dist\cncjs\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

D:\Xialiwei\github\cncjs\dist\cncjs\node_modules\@serialport\bindings>if not defined npm_config_node_gyp (node "D:\Applications\nvm\v1s\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "D:\Applications\nvmules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
锟斤拷止锟斤拷锟斤拷锟斤拷锟斤拷锟斤拷锟斤拷(Y/N)?

Error output:

Error output:
npm WARN deprecated bcrypt-nodejs@0.0.3: bcrypt-nodejs is no longer actively maintained. Please use bcrypt or bcryptjs. See https://ginode.bcrypt.js/wiki/bcrypt-vs-brypt.js to learn more about these two options
npm WARN deprecated @babel/polyfill@7.4.4: ???? This package has been deprecated in favor of separate inclusion of a polyfill and regehen needed). See the @babel/polyfill docs (https://babeljs.io/docs/en/babel-polyfill) for more information.
npm WARN deprecated core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Pleaependencies to the actual version of core-js@3.
npm WARN deprecated mkdirp@0.3.0: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API s to use Promises in 1.x.)
prebuild-install WARN install No prebuilt binaries found (target=10.0.0 runtime=electron arch=x64 libc= platform=win32)
^C^Cnpm ERR! code ELIFECYCLE
npm ERR! errno 3221225786
npm ERR! @serialport/bindings@2.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 3221225786
npm ERR!
npm ERR! Failed at the @serialport/bindings@2.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\MSI-PC\AppData\Roaming\npm-cache\_logs\2020-08-25T07_24_45_747Z-debug.log
  stackTrace=
```

raininja, 2020-09-23:
I'm experiencing this same issue.

LongLiveCHIEF, 2020-09-25:
This isn't a problem with cncjs, it's a problem with serialport.

LongLiveCHIEF, 2020-09-26:
So I've been reading up on serialport, since i had the same issue on my linux machines. You need to have python2 installed and on the PATH of the user who is running the `npm install`.  You'll also want to make sure you have a C compiler on your machine. For windows I think that would be a Visual C++ redistributable.

Adding those libs and such should allow the post-install for @serialport/bindings to run compile during its post-install steps.

@cheton neither Windows nor Linux install python2 by default anymore, so I'm guessing there will be more and more reports of this.  Should we add a note to the README about making sure you have python2 and a c compiler toolchain on your path in order for serialport to install?  The current note just says not to use `sudo`,.

MitchBradley, 2020-09-26:
Feel free to edit or create wiki pages with build instructions that you have personally tested.  One trick aspect of such testing is making sure that you are actually starting from scratch.

LongLiveCHIEF, 2020-09-27:
@MitchBradley I might do just that. 

Starting from scratch is actually pretty easy for me. I'm the octoprint-docker maintainer, so I have dozens of systems and a program I created called [bootcc](https://github.com/HackerHappyHour/bootcc) that allows me to quickly build a fresh machine from any distro that uses cloud-init. (I mainly test with Ubuntu x64 and 32).

Right now I'm just trying to distill it down to the shortest possible instructions, and I'm also providing feedback to node-serialport via https://github.com/serialport/node-serialport/issues/2109#issuecomment-699642350 about potential updates to their docs on compiling.

Windows has python3 in the Marketplace, and an easy command to install it (gotta look it up), and Ubunut and many other systems have python3 now installed by default but not callable as `python`.

Theoretically all you need to do is run an `update-alternatives` command so python 3 can be called using `python` by node-gyp, and you're good to go.  On windows it should be as easy as typing [python in the terminal](https://devblogs.microsoft.com/python/python-in-the-windows-10-may-2019-update/) to be taken to the store to set up python.

I haven't tested the simpler instruction sets for python 3 and node-gyp yet, but i did test with python 2 from a fresh Ubuntu 20.04 arm64 machine, where I ran:

```bash
# update and install c toolchain and python2
$ sudo apt update -y
$ sudo apt install -y build-essential gcc g++ python2

# verify python 2 and 3 are installed
$ ls -al /usr/bin/python*
lrwxrwxrwx  1 root root              7 Apr 15 05:45 /usr/bin/python -> python2
lrwxrwxrwx  1 root root              9 Mar 13  2020 /usr/bin/python2 -> python2.7
-rwxr-xr-x     1 root root 3694632 Apr    7 07:05 /usr/bin/python2.7
lrwxrwxrwx  1 root root              9 Mar 13  2020 /usr/bin/python3 -> python3.8
-rwxr-xr-x     1 root root 5453504   Jul 16 09:00 /usr/bin/python3.8

# verify system doesn't have an alternative for `python` already configured, which is used by node-gyp to compile serialport
$ sudo update-alternatives --list python
update-alternatives: error: no alternatives for python

# set system alternatives for python command
$ sudo update-alternatives --install /usr/bin/python python /usr/bin/python2 1
$ sudo update-alternatives --install /usr/bin/python python /usr/bin/python3 2
```
... and now serialport will compile and cncjs will install without errors.   

I've repeated that install for fresh Ubuntu installs for both arm64 and armhf architectures and observed success after first getting failures prior to those steps.

Unfortunately i have a _ton_ of docs to write for octoprint-docker, and they're going to take priority, so I can't take the time at the moment to do enough testing for this to have confidence in adding that doc to the wiki.

Hopefully this will help someone else with that time capacity to finish what I started, (or it will be here for posterity for when I do get time).

:beers:

MitchBradley, 2020-09-27:
> Unfortunately i have a _ton_ of docs to write for octoprint-docker, and they're going to take priority, so I can't take the time at the moment to do enough testing for this to have confidence in [...]

And thus we see the core problem: The intersection of "people with the necessary advanced knowledge and appropriate compute resources" and "people with time capacity" is the empty set. Anyone in the first set is virtually guaranteed to be overcommitted.

> On windows it should be as easy as typing python in the terminal to be taken to the store to set up python.

Setting up Python on Windows is easy, but managing python versions is not.  There are tools to do so, for example pyenv-win, but they introduce yet another layer of complexity that can be completely mind-blowing for a novice who is already overwhelmed with layer upon layer of unfamiliar stuff.

Actually, when I said "starting from scratch", I understated the problem.  Scratch is a reasonable starting point for a Raspberry Pi where you can just copy an OS image onto an SD card and proceed from there, under the assumption that you CNC Pi is probably not already in service.  On Windows, the machine probably is already in service, with dozens of other apps installed, so taking it down to newborn state is not practical.  So in addition to dealing with what you have to add, you also have to deal with other conflicting things might already be present.  Plus you have the can of worms of installing a C compiler plus maybe some ancillary tools - or dealing with one that is already installed that might be the wrong version.

Docker is the answer for some people, but it is one of those "other layers of complexity", and a particularly imposing one at that.  And if you are going to run Linux inside a Windows host, you have just introduced "all of Linux" to the complexity set. Not to mention the tricky aspects of accessing the host filesystem from inside the container.  The number of "minor problems" quickly becomes overwhelming.  Many people will give up long before a final solution.

The long-term solution that I like best has to do with entirely avoiding C compilation.  The Javascript parts of CNCjs are entirely machine-independent.  I have successfully built CNCjs on one machine, copied everything to a Raspberry Pi Zero, and then replaced the one binary file for serialport support.  That suggests that it should be possible to have a serialport-node module that contains multiple binaries to support all of the usual platforms - Win, Mac, Linux-x64, arm7, arm7hf.  With that, there would be no need to run the node-gyp step.

The crux of this matter digs down to the way that node-serialport handles binaries.  It would be relatively easy (for a suitable developer who has time - but see the first paragraph) to change node-serialport to use prebuildify.  See https://github.com/serialport/node-serialport/issues/975