cncjs/cncjs #645: Find out which endstop is being triggered?

Di3mex, 2020-09-27:
#### Description

I am in the process of upgrading/building my CNC machine. I have an issue were I am getting a triggered hardware endstop when I switch the spindle on. I would like to know which of the endstops x/y/z caused the message.
I suspect that vibrations are triggering one of the endstops. I just need t know which one.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

neilferreri, 2020-09-27:
The false limit trigger is most likely caused by electrical noise. Disable hard limits until you get that sorted out.

karoria, 2020-09-28:
@neilferreri is right. Most likely cause is electrical noise. I had the same problem in past. Adding a 4.7uF capacitor across limit pin (pcb input side) has solved my problem permanently. Yet, better way is to design a PCB with optoisolated inputs. All the best.

Di3mex, 2020-10-12:
Adding 0.47 uF caps resolved the issue. Thanks :-) !

Di3mex, 2020-10-27:
Even though my issue is solved I would still like to be able to see which endstop is being triggered. Today I accidentally triggered my Z endstop and I didn't realise it got triggered. 
Is this something I need to activate in GRBL or CNCjs? 
A simple "Z hardware endstop triggered" message would have saved me time.

Fusseldieb, 2022-05-09:
You can debug your endstops, at least with Marlin, using the command `M119` repeatedly.

MitchBradley, 2022-05-09:
The question was about Grbl, so the Marlin answer is not exactly apropos.

If you send the character ? in the cncjs console, the status response should contain a Pn: field that shows the pin states. You can learn more about this by reading Grbl documentation. The feature is part of Grbl, not cncjs