cncjs/cncjs #647: PCB copper removal tool path not correct

jponko, 2020-09-30:
#### Description

I can't get the copper removal gcode job created in FlatCAM 8.5 to work in CNCjs. If I load a gcode.nc file and zoom in to look at the tool path used to mill a PCB in CNCjs it's not the same as a displayed  in bCNC. If I start the CNC job it moves the tool in mini steps that lifts the mill head on each cut. This means it can take hours to mill a single track. The CNC mill is a Elekmil with grbl-1.1h.

The same job loaded in bCNC has a tool path that is a continuous cut

Screen shots of the tool paths for a pad for both CNCjs and bCNC are attached as well as the gcode file.
Why is the same gcode file interpreted differently?

#### Versions

- CNCjs: 1.9.22
- Node.js: 6.x
- NPM: 6.9.0

[jponko.zip](https://github.com/cncjs/cncjs/files/5302523/jponko.zip)

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux Mint 20.04

MitchBradley, 2020-09-30:
I can't reproduce the problem.  Your zoomed-in views lack the context necessary to determine which pad is involved.  I found one that seems similar and got this visualization, which seems correct, at coordinates near -15, -10.
[image]

jponko, 2020-10-01:
It wouldn't matter what pad or trace is viewed because all traces in the file are generated with  a standard path. It's the same file I tried to upload to CNCjs but kept displaying it as a seed based tool pah. My guess is that the web browser kept loading a cached version of the seed based file to the web server even after I deleted it and replaced it with a file with the same name but using the standard tool path. 

At the moment I'm milling a board using the standard tool path with bCNC. I'll try the files with CNCjs and  let you know how it goes. I'm not even going to try a seed based tool path again because it's way too slow..

jponko, 2020-10-02:
I  wasn't able to reproduce it so I'm going to close the issue.

closed 2020-10-02 by jponko