cncjs/cncjs #650: Hangs at long comment line

EvilHacker, 2020-10-04:
#### Description

Running g-code program just stops at what appears to be a very long comment line (generated output from Fusion 360).

The program begins with:
```
(1001)
(Machine)
(  vendor: Sienci Labs)
(  model: LongMill 30x30)
(  description: LongMill 30x30)
(T3  D=6.2 CR=0 - ZMIN=-5.1 - flat end mill)
G90 G94
G17
G21
(When using Fusion 360 for Personal Use, the feedrate of rapid moves is reduced to match the feedrate of cutting moves, which can increase machining time. Unrestricted rapid moves are available with a Fusion 360 Subscription.)

(2D Contour2)
(... rest of program ...)
```

When the LONG comment line is manually remove, the program runs fine.

I'm guessing there is an issue handling long lines and/or long comment lines.

#### Versions

Mac App "About": Version 1.9.22 (1.9.22.3668)

#### How Do You Install CNCjs?

- [x] Download the CNCjs Mac Application

#### CNC Controller

- [x] Grbl (Sienci LongBoard for LongMill)

#### Hardware

- [x] MacBook Pro

#### Operating System

- [x] Mac OS Catalina 10.15.6 (19G2021)

[1001.nc.txt](https://github.com/cncjs/cncjs/files/5324520/1001.nc.txt)

MitchBradley, 2020-10-05:
The whole GRBL community is up in arms about this. There are many complaints on Fusion 360 forums.  It isn't just CNCjs; other senders are having problems, because there is a fundamental 127-byte buffer size limit in classic GRBL. When running on AVR, as it usually does, the amount of RAM memory is quite limited so buffer sizes must be carefully controlled.  It is not easy, and sometimes not possible, to simply increase the buffer size.  Today a patch to Grbl_Esp32 was offered, increasing the buffer size to 255, which works around the problem.  Grbl_Esp32 runs on a 32-bit processor so the memory size is not so severely limited.
Changing all of the senders and controllers is probably not possible because there are so many of them out in the world, many of them used by people without the skill to upgrade. The better approach is for everyone affected to complain loudly to Autodesk for causing so much grief.

neilferreri, 2020-10-05:
What post processor?
You might try commenting out or removing the message command.
```
function onComment(message) {
  writeComment(message);
}
```

EvilHacker, 2020-10-05:
I was using the grbl post processor that comes with Fusion.

Autodesk is aware of this problem and is fixing it at their end in Fusion.
According to Autodesk, there is an 70 character line limit in GRBL.
[https://forums.autodesk.com/t5/fusion-360-support/personal-use-version-takes-away-rapid-moves-and-makes-my-carbide/m-p/9783976/highlight/true#M101832](https://forums.autodesk.com/t5/fusion-360-support/personal-use-version-takes-away-rapid-moves-and-makes-my-carbide/m-p/9783976/highlight/true#M101832)

Ideally, this problem should be fixed in GRBL.

I'm closing this issue as it is not a CNCjs problem.
Sorry for the distraction.

closed 2020-10-05 by EvilHacker

neilferreri, 2020-10-05:
Why would you need to send a string that long to a controller?

While waiting for Autodesk to fix it, modify the post processor so it's not there.
Then complain to Autodesk about removing the basic CAM functionalities that they claim won't affect hobby users.

MitchBradley, 2020-10-05:
Another reply from Autodesk:
`This is also of utmost priority to Autodesk. It was certainly an unintended consequence that this message would cause GRBL controllers to spazz out. We are currently working through this bug and hope to have a hotfix sooner rather than later.`
Regarding fixing it in the post - so far I haven't been able to determine whether the problem is in the post, or deeper in the non-modifyable parts of Fusion 360.  One might expect that they would put it somewhere more devious, since changing posts is so simple.

MitchBradley, 2020-10-05:
Someone just told me that they changed something outside the post so you can't just hack the post to fix the problem.

neilferreri, 2020-10-05:
> Someone just told me that they changed something outside the post so you can't just hack the post to fix the problem.

Even just for that message? Why would they force a message?
I understand the changing to remove rapids and multiple tools would have to happen in the initial CAM stage, but I guessed you could control what makes it past the post processor.