cncjs/cncjs #651: Is a Pi Zero enough?

Di3mex, 2020-10-12:
#### Description

Is a Pi zero sufficient for running CNCjs? I am just asking because a Pi Zero is not recommended for Octoprint. However, I think there will be much less communication required to run CNCjs compared to Octoprint... 

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

karoria, 2020-10-12:
For functional purpose it is fully supported. There are so many people reported that it runs fine on pi zero. If you run larger program files, say more than 3MB, then the process of loading that program files requires resource rich hardware. In that case, I would suggest pi4 4gb or 8gb. However, I personally observed that cncjs runs great with desktop or laptop systems with higher memory nc files say up to 8MB or 10MB.

MitchBradley, 2020-10-12:
Pi Zero cannot run the full CNCjs UI with acceptable performance.  It can run the CNCjs server component reasonably well, using another computer as the display.

closed 2020-10-12 by MitchBradley