cncjs/cncjs #652: Desktop App static port

willmackey2nd, 2020-10-15:
#### Description

CNCjs socket port number changes on every launch. Is it possible to set a static port number for it via config file or some other way? Mainly annoyance with AutoLevel and other external addons.

#### Versions

- CNCjs: 1.9.22
- Node.js: 14.13.1

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2020-10-15:
You can set it from the command line as described in https://github.com/cncjs/cncjs/wiki/Using-Pendants#using-the-app-with-browsers-on-other-machines-like-phones

willmackey2nd, 2020-10-16:
Thanks @MitchBradley! I had missed this.

closed 2020-10-16 by willmackey2nd