cncjs/cncjs #658: Probe connected to Z endstop doesn't work for tool change macros

Di3mex, 2020-11-07:
#### Description

I am trying to get the "Initial Tool" and "New Tool" macros to work. I have a micro switch at a fixed location that I had planned to use for this. However once the switch is triggered grbl reports hardware endstop hit and the macro fails to complete.
Disabling hardware endstops disables the switch and cncjs doesnt register the hit anymore. I have a CNCshield and it doesn't have a dedicated probe port.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

Di3mex, 2020-11-07:
I have a temporary (read permanent) fix for this. 
My Z endstop and probe are wired in parallel on the Z-Max pin. The pin for probe is the ACL pin on the CNC shield. Because the Z-Min and Z-Max pin share a pin I connected a dupont cable from the positive pin of my Z-Min port to the SCL pin.
I have modified the scripts to disable hard limits after the first M0 of the script and reenable it at the end.

I would like to be able to reenable hard limits even if I abort the script, but am not sure if that is possible.

neilferreri, 2020-11-07:
The A5 pin is the Probe pin.  There should be no reason to disable hard limits to use a probe.
Do you actually use a Z-min limit switch?