cncjs/cncjs #659: Last command never received

elin05, 2020-11-13:
#### Description

The last command is never received, even though all are sent. As a result, I need to manually Pause and then Stop the run every time.

#### Versions

- CNCjs: 1.9.22
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [x] Mac
- [ ] Linux

MitchBradley, 2020-11-13:
Since you didn't supply information about the way that commands are sent and what the commands are, it is impossible to recreate and debug the problem.  But I have a guess.  Maybe the last line in the file does not have an end-of-line character.

elin05, 2020-11-13:
Here are the last few lines: 
`X23.533 Y46.639
X23.180 Y46.714
X-0.180 Y46.714
G0 Z3.000
G00 Z3.000
M05
M30
%`

And here is a screenshot of when CNCjs gets in that state. Note:

- The Pause button is active
- The status is Idle
- 93/93 sent, but 92/93 received

[image]

MitchBradley, 2020-11-13:
You could try removing the trailing %